node-usb/node-usb #215: "Module not found: Error: Can't resolve 'aws-sdk'"

fudgepop01, 2017-12-15:
Hello, I'm using [this](https://github.com/chentsulin/electron-react-boilerplate) electron + react boilerplate. I've been trying to install this package for awhile and make it work properly with react/electron but it just seems to always hang-up on this error...

Whenever I try to incorporate this module into my code I get an error along the lines of:
>ERROR in ./node_modules/node-pre-gyp/lib/info.js
Module not found: Error: Can't resolve 'aws-sdk' in '/Users/Dominick/Creative Cloud Files/SVG_Builder/input-display/node_modules/node
-pre-gyp/lib'
 @ ./node_modules/node-pre-gyp/lib/info.js 14:14-32
 @ ./node_modules/node-pre-gyp/lib ^\.\/.*$
 @ ./node_modules/node-pre-gyp/lib/node-pre-gyp.js
 @ ./node_modules/usb/usb.js
 @ ./app/actions/display.js
 @ ./app/store/configureStore.dev.js
 @ ./app/store/configureStore.js
 @ ./app/index.js
 @ multi (webpack)-dev-server/client?http://localhost:1212 webpack/hot/dev-server react-hot-loader/patch webpack-dev-server/client?ht
tp://localhost:1212/ webpack/hot/only-dev-server ./app/index.js

>ERROR in ./node_modules/node-pre-gyp/lib/publish.js
Module not found: Error: Can't resolve 'aws-sdk' in '/Users/Dominick/Creative Cloud Files/SVG_Builder/input-display/node_modules/node
-pre-gyp/lib'
 @ ./node_modules/node-pre-gyp/lib/publish.js 17:14-32
 @ ./node_modules/node-pre-gyp/lib ^\.\/.*$
 @ ./node_modules/node-pre-gyp/lib/node-pre-gyp.js
 @ ./node_modules/usb/usb.js
 @ ./app/actions/display.js
 @ ./app/store/configureStore.dev.js
 @ ./app/store/configureStore.js
 @ ./app/index.js
 @ multi (webpack)-dev-server/client?http://localhost:1212 webpack/hot/dev-server react-hot-loader/patch webpack-dev-server/client?ht
tp://localhost:1212/ webpack/hot/only-dev-server ./app/index.js

>ERROR in ./node_modules/node-pre-gyp/lib/unpublish.js
Module not found: Error: Can't resolve 'aws-sdk' in '/Users/Dominick/Creative Cloud Files/SVG_Builder/input-display/node_modules/node
-pre-gyp/lib'
 @ ./node_modules/node-pre-gyp/lib/unpublish.js 15:14-32
 @ ./node_modules/node-pre-gyp/lib ^\.\/.*$
 @ ./node_modules/node-pre-gyp/lib/node-pre-gyp.js
 @ ./node_modules/usb/usb.js
@ ./app/actions/display.js
 @ ./app/store/configureStore.dev.js
 @ ./app/store/configureStore.js
 @ ./app/index.js
 @ multi (webpack)-dev-server/client?http://localhost:1212 webpack/hot/dev-server react-hot-loader/patch webpack-dev-server/client?ht
tp://localhost:1212/ webpack/hot/only-dev-server ./app/index.js

I've tried multiple ways of installing it and multiple different methods of using it with react (both the standard 'create-react-app' version and the boilerplate version mentioned above). For some reason though, no matter what I do it seems I can't make this usb module play nice with react. 

Can someone help? I'm a bit stuck on my program at the moment without this usb module...

fudgepop01, 2017-12-15:
so it turns out I misread the directions and installed it in the wrong package.json.

For the past 12 hours that was the issue. I now know what it's like to be a developer.

Apologies for any possible inconvenience ^^'

action-hong, 2018-02-06:
@dar2355  i use nw.js + webpack  and have some problem , how do you solve it? can you help me ?

lalitmee, 2018-02-17:
@dar2355 hey I am also facing the same error. Can you tell me how you solved the issue?

martintan, 2018-10-24:
Hi, I solved the problem by adding `externals` property in my `webpack.config.dev.js`
Like this:
```javascript
  externals: {
    "node-hid": 'commonjs node-hid',
    "usb": 'commonjs usb'
  }
```

jrobeson, 2019-03-19:
Fixed in https://github.com/tessel/node-usb/pull/300

closed 2019-03-19 by kevinmehall