node-usb/node-usb #220: crash electron when the computer sleep then wakeup in windows 10 os

hldgaofeng, 2017-12-28:
will crash electron when the computer sleep then wakeup in windows 10 os

Gandzjur, 2018-01-05:
I have the same problem even without using electron.
To reproduce I have installed usb@1.4.0 and run in node `let usb = require('usb')`
After computer wake up the node process crashed with "Access violation"

DarylBerryhill, 2018-01-08:
I have had that problem too. Its caused when the USB goes into standby mode when the computer sleeps.  If the usb library is polling it will either crash or lockup. The fix is to check for data error event then stop polling to prevent from crashing. 

```
device.interfaces[0].endpoints[0].on('error', function(error)
        {
            // try to stop polling
            device.interfaces[0].endpoints[0].stopPoll( function()
            {
                console.info(`polling has stopped `);
            });
        });
```

tzvetelin-vassilev, 2018-11-21:
I also experience the same problem With usb 1.3.3 I have found this:

[invalid-read.txt](https://github.com/tessel/node-usb/files/2604157/invalid-read.txt)

vshymanskyy, 2019-03-01:
I needed only usb insert/remove notifications, so switched to `usb-detect`. Works fine!

pedrobertao, 2019-09-18:
Still happening on 1.6.0. Any news on it ?

pedrobertao, 2019-09-18:
This module is causing the crash for Windows OS. Do we have an update on it ? It has been over two years already.

DarylBerryhill, 2019-09-18:
I'm using for over a year, USB 1.3.1 with Electron 1.8.1 with no problems at all running on the latest Windows 10. What exactly is your error? I may be able to can help.   for reference here is my stack.

    ],
    "dependencies": {
        "electron-debug": "^1.5.0",
        "file-saver": "^2.0.1",
        "serialport": "^6.2.2",
        "tabit": "^2.2.2",
        "usb": "^1.3.1",
        "winreg": "^1.2.4"
    },
    "devDependencies": {
        "@types/file-saver": "^2.0.0",
        "electron": "^1.8.1",
        "electron-rebuild": "^1.7.3",
        "json-server": "^0.12.1"
    }

Daryl

DarylBerryhill, 2019-09-18:
found another recent project where I'm using USB 1.6.0; SERIALPORT 7.1.5; ELECTRON 6.0.0. and successfully moving data from USB to serial simultaneously. Would you like for me to post my code? just offering a little bit of help.   again for reference here is my dependancy stack

  "dependencies": {
        "auto-launch": "^5.0.5",
        "serialport": "^7.1.5",
        "usb": "^1.6.0",
        "windows-shortcuts-appid": "^0.3.2"
  },
  "devDependencies": {
        "electron": "^6.0.0",
        "electron-builder": "^21.2.0",
        "electron-rebuild": "^1.8.5",
        "prebuild": "^9.0.1"
  },

pedrobertao, 2019-09-18:
Hello @DarylBerryhill , 

To reproduce I have installed usb@1.6.0 and after comming back from 'Sleep' state the screen goes white and no error reported. 
[Here](https://github.com/LedgerHQ/ledgerjs/issues/401) I demonstraded how to reproduce it 

We have used electron-react-boilerplate and the basic setup: 

"electron": "4.1.4" - Not the latest because of our db library. 
"@ledgerhq/hw-transport": "^4.66.0"
"@ledgerhq/hw-transport-node-hid": "^4.66.0"

@ledgehq has the usb 1.6.0

EnderAdel, 2020-09-08:
It's still happening. The rendering process keeps crashing. (I'm on version 10.1.1)
```
app.on('renderer-process-crashed', function(){
      notifier.notify({
      title: '[title]',
      message: 'The renderer process has crashed!'
    });
});
```