node-usb/node-usb #223: How read usb data with event?

ShuiShengYe, 2018-01-04:
I want to read usb data with event. When data received, the event is fired.

DarylBerryhill, 2018-01-08:
you can use the "on(data)" event to capture the data buffer. The data buffer needs to be converted to an array in order to use. 

here is an example to read 8 bytes from the USB.
usbLib    = require("usb");

device= usbLib.findByIds(VENDOR_ID, PRODUCT_ID);  // Find usb device by ID
device.open(true);                                                          // open device
device.interfaces[0].claim();                                            // claim interface

device.interfaces[0].endpoints[0].startPoll(1,8);             // start polling the USB for data event to fire

// when new data comes in a data event will be fired on the receive endpoint
device.interfaces[0].endpoints[0].on("data", function(dataBuf)
{
   let dataArr = Array.prototype.slice.call(new Uint8Array(dataBuf, 0, 8)); // convert buffer to array
   console.log( `data byte 3 is ${dataArr[3]}` );               // print data byte 3
}
 
I'm no expert but have been using this USB library with electron for over an year with good success. any questions post here or shot me an email  djberry(at)hiwaay(dot)net
Daryl

ShuiShengYe, 2018-01-09:
@DarylBerryhill Thank you for your answer. I tried it. It  loopped to receive data, but get a Buffer with noting.

DarylBerryhill, 2018-01-10:
This looks correct. The console log indicates a buffer was received. It just needs to be converted to an array. 
One thing to keep in mind. The previous example I gave`  ...startPoll(1,8) `The  **8** is the number of data bytes to receive.  When you convert the incoming USB buffer to array you need to know how many bytes the HID USB host is sending. Usually 64 bytes but could be less.

jhondavila, 2018-05-18:
I have a Code-39 scanner reader, I am trying to convert the data of a chain captured by the scanner, but I do not know how to do it. The data return event returns the following. I add my code below

`

var usb = require('usb'),
    term = usb.findByIds(11734, 417);

term.open();

var endpoints = term.interfaces[0].endpoints,
    inEndpoint = endpoints[0],
    outEndpoint = endpoints[1];

// term.claim()
// console.log(endpoints)
inEndpoint.transferType = 2;
// console.log(inEndpoint)
term.interfaces[0].claim()

inEndpoint.startPoll(1, 8);

var id = "", chunk;

var Buffers = require('buffers');
var bufs = Buffers();

inEndpoint.on('data', function (data) {
    console.log("**********")
    console.log(data)

    /* <Buffer 00 00 53 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 53 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 21 00 00 00 00 00>
<Buffer 00 00 25 00 00 00 00 00>
<Buffer 00 00 27 00 00 00 00 00>
<Buffer 00 00 21 00 00 00 00 00>
<Buffer 00 00 22 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 22 00 00 00 00 00>
<Buffer 00 00 25 00 00 00 00 00>
<Buffer 00 00 20 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00> */

});
inEndpoint.on('error', function (error) {
    console.log("on error", error);
});

inEndpoint.on('end', function () {
    console.log("on end");
});
outEndpoint.on("data", function (data) {
    console.log(data)

    console.log("-->", bufferOne.toString('utf8'))
});

outEndpoint.on('error', function (error) {
    console.log(error);
});`

kevinmehall, 2018-06-09:
It looks like your device is emulating a HID keyboard, so `data[2]` is the scancode of a keyboard key:
https://docs.mbed.com/docs/ble-hid/en/latest/api/md_doc_HID.html
http://www.usb.org/developers/hidpage/Hut1_12v2.pdf page 53

closed 2018-06-09 by kevinmehall