node-usb/node-usb #225: How to integrate with Angular 5

DarylBerryhill, 2018-01-08:
I've have successfully used this USB library with the latest version of Electron 1.8.3-beta-3 on windows 10.
I have successfully integrated Electron with Angular 5 framework and packaged to an .exe file. 

My question is how to integrate this USB library into Angular 5? 
  tried `import * as usbLib from 'usb'` // says usbLib doesn't exist. Of course I have installed it using npm and rebuilt it to match electron node version 

any ideas

theCrius, 2018-04-17:
I'm trying to do the same thing but I think the design of the application should be that you create hooks in the main.js and then allow angular to access those hooks.

as example:

main.js
```javascript
app.on('ready', () => {
  createWindow();
  mainWindow.webContents.on('did-finish-load', () => {
    mainWindow.webContents.send('usbDevices', usb.getDeviceList());
  });
});
```

and on angular, let's say app.component

```javascript
import { Component, OnInit } from '@angular/core';
const ipcRenderer = (<any>window).require('electron').ipcRenderer;

@Component({
  selector: 'app-root',
  templateUrl: './app.component.html',
  styleUrls: ['./app.component.css']
})

export class AppComponent implements OnInit {

  constructor() { }

  getUsbDevices() {
    ipcRenderer.on('usbDevices', (event, message) => {
      console.log(message);
    });
  }

  ngOnInit() {
    this.getUsbDevices();
  }
}
```

You can specify custom hooks in main.js and have them trigger an action on request as well.
Check electron docs for more info.

semireg, 2019-04-18:
Hi. Unfortunately, this isn't a question for this project. You should find an angular-focused group to help you.

I'm using this library in an Electron/React app. I used this project: https://github.com/electron-react-boilerplate/electron-react-boilerplate. All USB work should be done in main.js. For my use case, I use Immutable records to represent my USB devices and serialize them across IPC. Within each renderer, I use React/Redux to maintain state.

Please close this issue. Good luck!

closed 2019-05-19 by kevinmehall