node-usb/node-usb #228: usb@1.3.1 install /home/pi/node_modules/usb configure error Raspberrypi 3

Smart71, 2018-01-28:
i am stucked no idea what happened...

tried sudo npm install usb

and then get this...

pi@raspberrypi:~ $ sudo npm uninstall usb

> usb@1.3.1 install /home/pi/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.3.1/usb_bindings-v1.3.1-node-v59-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for usb@1.3.1 and node@9.4.0 (node-v59 ABI, glibc) (falling back to source compile with node-gyp) 
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/pi/node_modules/usb/build'
gyp ERR! System Linux 4.9.77-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "--fallback-to-build" "--module=/home/pi/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/home/pi/node_modules/usb/src/binding"
gyp ERR! cwd /home/pi/node_modules/usb
gyp ERR! node -v v9.4.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/home/pi/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/pi/node_modules/usb/src/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:160:13)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:943:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:220:5)
node-pre-gyp ERR! System Linux 4.9.77-v7+
node-pre-gyp ERR! command "/usr/local/bin/node" "/home/pi/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/node_modules/usb
node-pre-gyp ERR! node -v v9.4.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.39
node-pre-gyp ERR! not ok 
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/home/pi/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/pi/node_modules/usb/src/binding' (1)
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.3.1 (node_modules/usb):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.3.1 install: `node-pre-gyp install --fallback-to-build`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

up to date in 27.836s

kevinmehall, 2018-01-28:
Do not run NPM as root (https://github.com/nodejs/node-gyp/issues/454), and make sure you have the build dependencies (gcc, libudev-dev) installed.

Duplicate of #175

closed 2018-01-28 by kevinmehall

darghorn, 2018-02-09:
Hello,

What about the 404 error ? Does not seems normal.
I've got the same problem, and do not run my install with root

Regards