node-usb/node-usb #229: How to rebuild this package with the latest version of libusb on Windows?

jsho5150, 2018-01-30:
Version 1.0.17 of libusb which is included with this project appears to have an issue when used within a virtual machine.  The getDeviceList function only returns the USB hub object and not the children (usb devices) of that hub.  I downloaded and compiled version 1.0.22 of libusb and the issue appears to be resolved.  (I can get a list of the usb devices in my virtual machine.)  My question is, how can I get the updated version of libusb built into this project?  (I have a bleno based project which uses this package to communicate with a Bluetooth 4.0 USB dongle.)

jsho5150, 2018-02-01:
Update.  I was able to get this package to compile with version 1.0.22 of libusb by doing the following:

Set 'variables' in binding.gyp to:
  'variables': {
    'use_udev%': 1,
    'use_system_libusb%': 'false',
    'module_name': 'usb_bindings',
    'module_path': './src/binding'
  },

Added the following win 'sources' in libusb.gypi:
	    'libusb/libusb/os/windows_nt_common.c',
            'libusb/libusb/os/windows_nt_common.h',
	    'libusb/libusb/os/windows_nt_shared_types.h',
	    'libusb/libusb/os/windows_usbdk.c',
            'libusb/libusb/os/windows_usbdk.h',

Changed the following win 'sources' in libusb.gypi:
From:
            'libusb/libusb/os/windows_usb.c',
            'libusb/libusb/os/windows_usb.h',
To:
            'libusb/libusb/os/windows_winusb.c',
            'libusb/libusb/os/windows_winusb.h',

After the edits I ran: (I have VS 2015 Express installed and I believe the rebuild below used that to compile.)
  npm install -g node-gyp
  node-gyp rebuild 

So, with all of that done, my bleno based project can now see the children devices for my USB hub in the virtual machine. (VMWare Horizons)  Unfortunately, when my code tries to open that device, I am now greeted with a "LIBUSB_ERROR_NOT_SUPPORTED" error.  I moved a copy of the compiled 'usb_bindings.node' to a non-VM Windows PC and received the same error.  Does anyone have insight regarding how to resolve the 'LIBUSB_ERROR_NOT_SUPPORTED' error?  I'm thinking that I need to modify something else in this package and rebuild, but I'm not sure what.  (My best guess is something in 'node_usb.cc' or 'device.cc')

kevinmehall, 2018-02-01:
What driver do you have installed for the BLE dongle? To use libusb, you need the `winusb` driver bound to the device. You can install it with http://zadig.akeo.ie/.

jsho5150, 2018-02-01:
Yes, I should have mentioned that in my previous post.  I'm using winsub that was installed via zadig.  I have tried reinstalling and still have the same "LIBUSB_ERROR_NOT_SUPPORTED" error in the Virtual Machine and native Windows 7.

kevinmehall, 2018-02-02:
Hmm, the only thing that comes to mind relating to the libusb upgrade is that the version of libusb that is shipped with this library by default is patched with support for hotplug events on windows, which is not in upstream libusb. I believe those APIs return LIBUSB_ERROR_NOT_SUPPORTED without the patch, but node-usb should only call them if you do a `usb.on('attach')`.

Is there a stack trace when this happens that shows which function encountered the error?

jsho5150, 2018-02-02:
This is what I see for output during the crash:

C:\VT_tools\Osiris\VT\TestFramework\BLEdeviceSim\node_modules\usb\usb.js:33
        this.__open()
             ^

Error: LIBUSB_ERROR_NOT_SUPPORTED
    at Error (native)
    at Device.usb.Device.open (C:\VT_tools\Osiris\VT\TestFramework\BLEdeviceSim\
node_modules\usb\usb.js:33:7)
    at BluetoothHciSocket.bindUser (C:\VT_tools\Osiris\VT\TestFramework\BLEdevic
eSim\node_modules\bluetooth-hci-socket\lib\usb.js:73:19)
    at BluetoothHciSocket.bindRaw (C:\VT_tools\Osiris\VT\TestFramework\BLEdevice
Sim\node_modules\bluetooth-hci-socket\lib\usb.js:28:8)
    at Hci.init (C:\VT_tools\Osiris\VT\TestFramework\BLEdeviceSim\NodeJsApplicat
ion\lib\hci-socket\hci.js:110:35)
    at BlenoBindings.init (C:\VT_tools\Osiris\VT\TestFramework\BLEdeviceSim\Node
JsApplication\lib\hci-socket\bindings.js:99:13)
    at Bleno.<anonymous> (C:\VT_tools\Osiris\VT\TestFramework\BLEdeviceSim\NodeJ
sApplication\lib\bleno.js:52:22)
    at emitTwo (events.js:106:13)
    at Bleno.emit (events.js:191:7)
    at _addListener (events.js:226:14)

jsho5150, 2018-02-02:
Quick update.  My build with libusb version 1.0.22 is working on native windows.  That said, even with the WinUSB driver installed, I continue to receive LIBUSB_ERROR_NOT_SUPPORTED in my virtual machine.

jrobeson, 2018-07-26:
I've found that the newest libusb seems to reduce hangs with my noble ble application.

closed 2021-11-08 by thegecko

GuilhermeCouto, 2023-08-04:
Hi, I am facing this problem, it prints but then it get FATAL ERROR:

FATAL ERROR:  LIBUSB_ERROR_NOT_SUPPORTED
 1: 00007FF70FB207BF node_api_throw_syntax_error+175823
 2: 00007FF70FAA5796 DSA_meth_get_flags+59654
 3: 00007FF70FAA758C node::OnFatalError+252
 4: 00007FF70FADE49C napi_fatal_error+156
 5: 00007FFAAFFB1887
 6: 00007FFAAFFF876E
 7: 00007FFAAFFDCDA0
 8: 00007FFAAFFDBCF9
 9: 00007FFB037F3646 RtlCaptureContext2+1190
10: 00007FFAAFFBD3E3
11: 00007FFAAFFC021B
12: 00007FFAAFFB94C4
13: 00007FF70FADB492 node_module_register+3506
14: 00007FF70FB803F0 uv_timer_stop+736
15: 00007FF70FB805C7 uv_timer_stop+1207
16: 00007FF70FB7CA3B uv_update_time+491
17: 00007FF70FB7C582 uv_run+1266
18: 00007FF70FB4EF25 node::SpinEventLoop+325
19: 00007FF70FA5D1B8 v8::internal::interpreter::BytecodeArrayWriter::constant_array_builder+149496
20: 00007FF70FAE22E2 node::InitializeOncePerProcess+2882
21: 00007FF70FAE3B13 node::Start+835
22: 00007FF70F8E7F2C AES_cbc_encrypt+124252
23: 00007FF710B63428 inflateValidate+18776
24: 00007FFB016426AD BaseThreadInitThunk+29
25: 00007FFB037AAA68 RtlUserThreadStart+40

anyone knows why?

PR #453 (2021-11-08): Libusb update to v1.0.23
This PR switches to use the upstream [libusb library](https://github.com/libusb/libusb) at `v1.0.23`.

Based on comments and feedback from @ivannkf, @koush and @tombertrand I wanted to see how broken this is as apparently it's just missing Windows hotplug support.

My testing shows that with this release, hotplugging works on Windows (?!), does anyone know what I'm missing or shall we switch to the upstream release?

If so, this fixes #229 and #161

I've released a test version for people to try and I'd love some feedback as this would be a major change:

```bash
npm i [... cut]