node-usb/node-usb #230: Build errors when rebuilding for electron(1.7.9)  on Win10-ia32

zhaiyuzhai, 2018-02-11:
I have built node-usb for electron on win10-x64 successfuly,but when i build for electron on win10-ia32 it comes the error just like that:

```
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp.cmd configure --fallback-to-build --
module=C:\Users\zhaiyu32\Desktop\BLE_DEMO(WIN32)\node_modules\usb\src\binding\usb_bindings.node
 --module_name=usb_bindings --module_path=C:\Users\zhaiyu32\Desktop\BLE_DEMO(WIN32)\node_module
s\usb\src\binding --python=python2.7 --msvs_version=2015' (255)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\zhaiyu32\Desktop\BLE_DEMO(WIN
32)\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:126:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:925:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:2
09:5)
node-pre-gyp ERR! System Windows_NT 10.0.16299
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\zhaiyu32\\Desktop\\
BLE_DEMO(WIN32)\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd C:\Users\zhaiyu32\Desktop\BLE_DEMO(WIN32)\node_modules\usb
node-pre-gyp ERR! node -v v8.9.4
node-pre-gyp ERR! node-pre-gyp -v v0.6.39
node-pre-gyp ERR! not ok
Failed to execute 'node-gyp.cmd configure --fallback-to-build --module=C:\Users\zhaiyu32\Deskto
p\BLE_DEMO(WIN32)\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --m
odule_path=C:\Users\zhaiyu32\Desktop\BLE_DEMO(WIN32)\node_modules\usb\src\binding --python=pyth
on2.7 --msvs_version=2015' (255)
```
can anyone help me?

lishichao1002, 2018-09-13:
+1

```
> usb@1.3.3 install Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb
> node-pre-gyp install --fallback-to-build

�ڴ˽��������һ������һ����Ŀ����Ҫ���ò������ɣ������ӡ�/m�����ء�
�ڴ˽��������һ������һ����Ŀ����Ҫ���ò������ɣ������ӡ�/m�����ء�
  core.c
  descriptor.c
  core.c
  descriptor.c
  hotplug.c
  hotplug.c
  io.c
  io.c
  strerror.c
  strerror.c
  sync.c
  sync.c
  poll_windows.c
  poll_windows.c
  threads_windows.c
  threads_windows.c
..\libusb\libusb\os\poll_windows.c(707): warning C4018: '<': signed/unsigned mismatch [Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb\build\libusb.vcxproj]
  windows_usb.c
..\libusb\libusb\os\poll_windows.c(707): warning C4018: '<': signed/unsigned mismatch [Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb\build\libusb.vcxproj]
  windows_usb.c
..\libusb\libusb\os\windows_usb.c(2558): warning C4018: '<': signed/unsigned mismatch [Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb\build\libusb.vcxproj]
..\libusb\libusb\os\windows_usb.c(2558): warning C4018: '<': signed/unsigned mismatch [Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb\build\libusb.vcxproj]
  win_delay_load_hook.cc
  win_delay_load_hook.cc
C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\src\win_delay_load_hook.cc : fatal error C1041: cannot open program database 'Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb\build\Release\obj\libusb\libusb.pdb'; if multiple CL.EXE write to the same .PDB file, please use /FS [Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb\build\libusb.vcxproj]
Failed to execute 'C:\Program Files\nodejs\node.exe C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb\src\binding --napi_version=3
--node_abi_napi=napi' (1)

Error output:
node-pre-gyp WARN Using request for node-pre-gyp https download
node-pre-gyp WARN Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.3.3/usb_bindings-v1.3.3-electron-v1.8-win32-ia32.tar.gz
node-pre-gyp WARN Pre-built binaries not found for usb@1.3.3 and electron@1.8.6 (electron-v1.8 ABI, unknown) (falling back to source compile with node-gyp)
node-pre-gyp WARN Pre-built binaries not installable for usb@1.3.3 and electron@1.8.6 (electron-v1.8 ABI, unknown) (falling back to source compile with node-gyp)
node-pre-gyp WARN Hit error Connection closed while downloading tarball file
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:238:12)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--module=Z:\\Users\\lishichao\\workspaces\\webstorm\\soda-b-pc\\dist\\electron\\node_modules\\usb\\src\\binding\\usb_bindings.node" "--module_name=usb_bindings" "--module_path=Z:\\Users\\lishichao\\workspaces\\webstorm\\soda-b-pc\\dist\\electron\\node_modules\\usb\\src\\binding" "--napi_version=3" "--node_abi_napi=napi"
gyp ERR! cwd Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb
gyp ERR! node -v v10.8.0
gyp ERR! node-gyp -v v3.7.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Program Files\nodejs\node.exe C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb\src\binding --napi_version=3 --node_abi_napi=napi' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:962:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:249:5)
node-pre-gyp ERR! System Windows_NT 6.1.7601
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "Z:\\Users\\lishichao\\workspaces\\webstorm\\soda-b-pc\\dist\\electron\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb
node-pre-gyp ERR! node -v v10.8.0
node-pre-gyp ERR! node-pre-gyp -v v0.10.3
node-pre-gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb@1.3.3 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
[... 29 lines cut ...]
[... cut ...]
    at Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\node_modules\electron-builder\src\builder.ts:310:40
    at Generator.next (<anonymous>)
From previous event:
    at _build (Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\node_modules\electron-builder\out\builder.js:61:21)
    at build (Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\node_modules\electron-builder\src\builder.ts:280:10)
    at electronBuilderProcess (Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\scripts\build.js:48:10)
```

glaucia86, 2018-10-24:
Same error here

`Error: npm.cmd exited with code 1
Output:

> @serialport/bindings@2.0.2 install D:\Labs\poc-electron-serial-port\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

> usb@1.3.3 install D:\Labs\poc-electron-serial-port\node_modules\usb
> node-pre-gyp install --fallback-to-build

Compilando os projetos desta solu��o um de cada vez. Para habilitar o build paralelo, adicione a op��o "/m".
MSBUILD : Configuration error MSB4148: O nome de uma propriedade armazenada na chave do Registro "HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\MSBuild\ToolsVersions\4.0" tem comprimento zero.
Failed to execute 'C:\Program Files\nodejs\node.exe C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=D:\Labs\poc-electron-serial-port\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=D:\Labs\poc-electron-serial-port\node_modules\usb\src\binding --napi_version=3 --node_abi_napi=napi' (1)                                                                                   
Error output:
node-pre-gyp WARN Using request for node-pre-gyp https download
node-pre-gyp WARN Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.3.3/usb_bindings-v1.3.3-electron-v1.8-win32-x64.tar.gz
node-pre-gyp WARN Pre-built binaries not found for usb@1.3.3 and electron@1.8.8 (electron-v1.8 ABI, unknown) (falling back to source compile with node-gyp)
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:262:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Windows_NT 10.0.16299
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--module=D:\\Labs\\poc-electron-serial-port\\node_modules\\usb\\src\\binding\\usb_bindings.node" "--module_name=usb_bindings" "--module_path=D:\\Labs\\poc-electron-serial-port\\node_modules\\usb\\src\\binding" "--napi_version=3" "--node_abi_napi=napi"
gyp ERR! cwd D:\Labs\poc-electron-serial-port\node_modules\usb
gyp ERR! node -v v8.12.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Program Files\nodejs\node.exe C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=D:\Labs\poc-electron-serial-port\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=D:\Labs\poc-electron-serial-port\node_modules\usb\src\binding --napi_version=3 --node_abi_napi=napi' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (D:\Labs\poc-electron-serial-port\node_modules\usb\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:126:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:915:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
node-pre-gyp ERR! System Windows_NT 10.0.16299
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "D:\\Labs\\poc-electron-serial-port\\node_modules\\usb\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd D:\Labs\poc-electron-serial-port\node_modules\usb
node-pre-gyp ERR! node -v v8.12.0
node-pre-gyp ERR! node-pre-gyp -v v0.10.3
node-pre-gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb@1.3.3 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the usb@1.3.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Glaucia Lemos\AppData\Roaming\npm-cache\_logs\2018-10-24T04_50_14_780Z-debug.log

    at ChildProcess.childProcess.once.code (C:\Users\Glaucia Lemos\AppData\Roaming\npm\node_modules\electron-builder\node_modules\builder-util\src\util.ts:254:14)
    at Object.onceWrapper (events.js:317:30)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at maybeClose (internal/child_process.js:915:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
From previous event:
    at rebuild (C:\Users\Glaucia Lemos\AppData\Roaming\npm\node_modules\electron-builder\node_modules\app-builder-lib\out\util\yarn.js:239:18)
    at C:\Users\Glaucia Lemos\AppData\Roaming\npm\node_modules\electron-builder\node_modules\app-builder-lib\src\util\yarn.ts:20:11
    at Generator.next (<anonymous>)
    at runCallback (timers.js:810:20)
    at tryOnImmediate (timers.js:768:5)
    at processImmediate [as _immediateCallback] (timers.js:745:5)
From previous event:
    at installOrRebuild (C:\Users\Glaucia Lemos\AppData\Roaming\npm\node_modules\electron-builder\node_modules\app-builder-lib\out\util\yarn.js:68:17)
[... cut ...]
    at Object.get [as argv] (C:\Users\Glaucia Lemos\AppData\Roaming\npm\node_modules\electron-builder\node_modules\yargs\yargs.js:1000:21)
    at Object.<anonymous> (C:\Users\Glaucia Lemos\AppData\Roaming\npm\node_modules\electron-builder\src\cli\cli.ts:25:28)
    at Module._compile (module.js:653:30)
    at Object.Module._extensions..js (module.js:664:10)
    at Module.load (module.js:566:32)
    at tryModuleLoad (module.js:506:12)
    at Function.Module._load (module.js:498:3)
    at Function.Module.runMain (module.js:694:10)
    at startup (bootstrap_node.js:204:16)
    at bootstrap_node.js:625:3`

kevinmehall, 2018-10-24:
These error messages don't include anything useful.

People have reported success using this module with Electron by following the instructions here:
https://github.com/electron/electron/blob/master/docs/tutorial/using-native-node-modules.md#using-native-node-modules

closed 2018-10-24 by kevinmehall