node-usb/node-usb #231: Does not work on latest Tessel2 firmware

necccc, 2018-02-18:
Trying to use this module on a tessel2, and it completely fails me.
Which is strange, because I have a running tessel2 with an older fw which works with usb@1.2.0 perfectly as I write these lines.

Environment:
```
node -v
v6.10.3
npm -v 
3.10.10
node-pre-gyp INFO t2-cli: 0.1.17
node-pre-gyp INFO t2-firmware: 0.1.0
node-pre-gyp INFO Node.js: 6.10.3
```

When I try to run a simple example on tessel, 

```
var usb = require('usb');
var list = usb.getDeviceList();

console.log(list);
```

I got the following error

```
module.js:471
    throw err;
    ^

Error: Cannot find module '/tmp/remote-script/node_modules/usb/src/binding/usb_bindings.node'
    at Function.Module._resolveFilename (module.js:469:15)
    at Function.Module._load (module.js:417:25)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/tmp/remote-script/node_modules/usb/usb.js:1:452)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
```

Additionally, here's the version info from the tessel, where the 1.2.0 works like a charm
```
node-pre-gyp INFO t2-cli: 0.1.17
node-pre-gyp INFO t2-firmware: 0.0.16
node-pre-gyp INFO Node.js: 4.5.0
```

rwaldron, 2018-02-26:
This module is not intended to be used _on_ the Tessel 2.

rwaldron, 2018-02-26:
> INFO t2-firmware: 0.0.16

I recommend running `t2 update` to get the latest firmware

closed 2018-02-26 by rwaldron

necccc, 2018-02-27:
Hi Rick,

that's a bit disappointing, since when I've first tried it a few years ago it worked perfectly, and was happy to find a node lib that exposes raw USB functionality.

What do you recommend then I could use instead of this one?

Looked around in the tessel module repos, if there is an undocumented usb api hidden somewhere but no luck. Seems the existing libs solve fixed usecases using native libs on the board, which kinda ties our hand - in this usecase, the fadecandy is a custom usb device, and was able to communicate with it by building up the connection via USB and sending the binary packets.

Any suggestion would be really appreciated, thanks!

rwaldron, 2018-02-28:
@kevinmehall is right, I forgot that we were indeed able to cross-compile `usb` for use on the Tessel 2. I've done that now and uploaded the pre-compiled binaries to the remote host.

kmegoryc, 2018-10-23:
@rwaldron how did you do that?