node-usb/node-usb #232: Prebuilt binary not found

lalitmee, 2018-02-20:
Hello, I am trying to use **node-usb** in my electron application with React. But I am facing issues.

- **OS :** Ubuntu 16.04
- **node version :** v9.3.0

- **error** 
```
node-pre-gyp ERR! Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.3.1/usb_bindings-v1.3.1-node-v59-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for usb@1.3.1 and node@9.3.0 (node-v59 ABI, glibc) (falling back to source compile with node-gyp) 
```
When I saw in this repo for this file which it is trying to download, I found that it doesn't exist here. Please help me. Thank you.

kevinmehall, 2018-02-20:
There are no pre-built binaries for Linux (or Electron on any platform). It looks like it successfully built from source automatically though.

closed 2018-02-20 by kevinmehall

lalitmee, 2018-02-21:
Actually, I found the solution. Thanx. I am sharing the link to the solution so that someone can get help. 
[Use node-usb inside electron](https://discuss.atom.io/t/use-node-usb-inside-electron/16748/3?u=lalitmee)