node-usb/node-usb #240: this.__getBosDescriptor() causing timeouts

joshberg, 2018-04-12:
I am trying to integrate with a USB scale. I have created and installed the WinUSB driver from Zadig because the vendor no longer provides the driver. When trying to open the the device, Electron would lock up. Breaking through the source code, I discovered that the usb library is timing out on this block of code:
```
Object.defineProperty(usb.Device.prototype, "bosDescriptor", {
	get: function() {
		try {
			return this._bosDescriptor || (this._bosDescriptor = this.__getBosDescriptor())
		} catch(e) {
			return null;
		}
	}
});
```

To temporarily resolve the issue, I commented out the function.

```
Object.defineProperty(usb.Device.prototype, "bosDescriptor", {
	get: function() {
		try {
			return this._bosDescriptor || (this._bosDescriptor = null) //this.__getBosDescriptor())
		} catch(e) {
			return null;
		}
	}
});
```

Should this be an asynchronous function call?

kevinmehall, 2018-04-12:
What version are you using? You want 1.3.1. That change was reverted on master and 1.4.0 was pulled due to that issue.

closed 2018-04-12 by kevinmehall

joshberg, 2018-04-12:
I have version 1.4.0. I will revert and try again.

kevinmehall, 2018-04-12:
Any idea how you ended up with 1.4.0? I set `latest` on npm back to 1.3.1, so I thought it wouldn't install 1.4.0 unless you explicitly put it in your package.json, but maybe I should release a 1.4.1 identical to 1.3.1.

joshberg, 2018-04-12:
Honestly, I am not sure. I believe I reinstalled the package yesterday through an npm update or npm install while troubleshooting Electron.