node-usb/node-usb #247: i cant' open device and occured error LIBUSB_ERROR_ACCESS

cofham, 2018-06-04:
`usb.on('attach', function(device) {
	device.open();
});`

but can't open device and error LIBUSB_ERROR_ACCESS 

how can i solve this problem?

grigdodon, 2018-06-20:
@cofham hi. did you manage to solve this issue? thanks!

KatherineGuerrero, 2018-08-03:
This always happens to me when I run the app without sudo, if I use sudo it works. Do you know how to give the app those permissions?

kevinmehall, 2018-08-03:
On Linux you can add udev rules to give a user or group access to certain USB devices:
http://ask.xmodulo.com/change-usb-device-permission-linux.html

closed 2021-01-30 by kevinmehall

juliusvaart, 2021-05-18:
Adding the MODE="0660" permissions and adding GROUP="plugdev" to all USB-devices fixed it for us on Ubuntu 20.04.

Add:
`SUBSYSTEM=="usb", MODE="0660", GROUP="plugdev"`

To:
`/etc/udev/rules.d/99-usb-permissions.rules`

Reload udev rules
`sudo udevadm control --reload-rules`

Reboot or replug USB-devices.