node-usb/node-usb #248: install failed with node 10

drom, 2018-06-09:
Trying to install on the latest Node 10:

```
drom@localhost:~/work/lab/ecp5> nvm i 10
v10.4.0 is already installed.
Now using node v10.4.0 (npm v6.1.0)
drom@localhost:~/work/lab/ecp5> npm i

> usb@1.3.1 install /home/drom/work/lab/ecp5/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.3.1/usb_bindings-v1.3.1-node-v64-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for usb@1.3.1 and node@10.4.0 (node-v64 ABI, glibc) (falling back to source compile with node-gyp) 
node-pre-gyp ERR! Tried to download(undefined): https://github.com/tessel/node-usb/releases/download/1.3.1/usb_bindings-v1.3.1-node-v64-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for usb@1.3.1 and node@10.4.0 (node-v64 ABI, glibc) (falling back to source compile with node-gyp) 
make: Entering directory '/home/drom/work/lab/ecp5/node_modules/usb/build'
make: *** No targets.  Stop.
make: Leaving directory '/home/drom/work/lab/ecp5/node_modules/usb/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/drom/.nvm/versions/node/v10.4.0/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:237:12)
gyp ERR! System Linux 4.16.12-2-default
gyp ERR! command "/home/drom/.nvm/versions/node/v10.4.0/bin/node" "/home/drom/.nvm/versions/node/v10.4.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/drom/work/lab/ecp5/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/home/drom/work/lab/ecp5/node_modules/usb/src/binding"
gyp ERR! cwd /home/drom/work/lab/ecp5/node_modules/usb
gyp ERR! node -v v10.4.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/home/drom/.nvm/versions/node/v10.4.0/bin/node /home/drom/.nvm/versions/node/v10.4.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/drom/work/lab/ecp5/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/drom/work/lab/ecp5/node_modules/usb/src/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/drom/work/lab/ecp5/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:961:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
node-pre-gyp ERR! System Linux 4.16.12-2-default
node-pre-gyp ERR! command "/home/drom/.nvm/versions/node/v10.4.0/bin/node" "/home/drom/work/lab/ecp5/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/drom/work/lab/ecp5/node_modules/usb
node-pre-gyp ERR! node -v v10.4.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.39
node-pre-gyp ERR! not ok 
Failed to execute '/home/drom/.nvm/versions/node/v10.4.0/bin/node /home/drom/.nvm/versions/node/v10.4.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/drom/work/lab/ecp5/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/drom/work/lab/ecp5/node_modules/usb/src/binding' (1)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb@1.3.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb@1.3.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/drom/.npm/_logs/2018-06-09T20_03_21_997Z-debug.log
```
[2018-06-09T20_03_21_997Z-debug.log](https://github.com/tessel/node-usb/files/2087292/2018-06-09T20_03_21_997Z-debug.log)

kevinmehall, 2018-06-09:
It builds for me on Node 10.4.0. It looks like you were potentially missing files from the build directory / the gyp step got skipped somehow.

mischnic, 2018-06-10:
I have this problem as well (macOS), it works with a newer node-pre-gyp version (#244).

> It builds for me on Node 10.4.0. It looks like you were potentially missing files from the build directory / the gyp step got skipped somehow.

With the currently used version, everything runs twice in parallel (and one deletes a file the other needs):

> ```
>node-pre-gyp ERR! Tried to download(404): https://github.com/tessel/node->usb/releases/download/1.3.1/usb_bindings-v1.3.1-node-v64-linux-x64.tar.gz 
>node-pre-gyp ERR! Pre-built binaries not found for usb@1.3.1 and node@10.4.0 (node-v64 ABI, glibc) >(falling back to source compile with node-gyp) 
>node-pre-gyp ERR! Tried to download(undefined): https://github.com/tessel/node->usb/releases/download/1.3.1/usb_bindings-v1.3.1-node-v64-linux-x64.tar.gz 
>node-pre-gyp ERR! Pre-built binaries not found for usb@1.3.1 and node@10.4.0 (node-v64 ABI, glibc) >(falling back to source compile with node-gyp) 
>```

kevinmehall, 2018-06-10:
Released 1.3.2 with the node-pre-gyp upgrade as binary builds for Node 10 on Windows and macOS. Tested on Linux and confirmed that the node-pre-gyp upgrade solves the double-build issue.

closed 2018-06-10 by kevinmehall

drom, 2018-06-11:
Great. It works!

drom, 2018-07-09:
Build keep randomly failing on Node 10.6.0
Here is an example: https://gitlab.com/drom/jtag/-/jobs/80286481

kevinmehall, 2018-07-10:
The node-gyp upgrade didn't fix the double-build problem, which is https://github.com/mapbox/node-pre-gyp/issues/391

kevinmehall, 2018-07-10:
As a workaround, `npm install --build-from-source=usb` might skip the network request whose error handling causes that issue.

drom, 2018-07-10:
it still randomly failing even with the option.
https://gitlab.com/drom/jtag/-/jobs/80770415

kevinmehall, 2018-07-10:
```
$ node --version --build-from-source=usb
v10.6.0
$ npm install
> usb@1.3.2 install /builds/drom/jtag/node_modules/usb
```
I think you added the argument to the wrong command.

drom, 2018-07-10:
Yes, you are right. The workaround works.