node-usb/node-usb #249: Polling Start/Stop on Input Endpoint

drom, 2018-06-16:
Hello,

I have some trouble understanding how Polling Start / Stop mechanism works.

```js
const handler = dat => {
    console.log(dat);
    epIn.stopPoll(() => {
        epIn.startPoll(1);
    });
};
epIn.on('data', handler);
epIn.startPoll(1);
```

  * attaching event handler `.on('data', ...)`
  * starting polling `.startPoll(1)`
  * getting callback with the data
  * stopping polling `.stopPoll(cb)` with callback to re-enable polling.
  * and getting Error:

```js
/home/drom/work/lab/drom/ecp5/node_modules/usb/usb.js:268
                throw new Error("Polling already active")
                ^

Error: Polling already active
    at InEndpoint.Endpoint.startPoll (/home/drom/work/lab/drom/ecp5/node_modules/usb/usb.js:268:9)
    at InEndpoint.startPoll (/home/drom/work/lab/drom/ecp5/node_modules/usb/usb.js:324:61)
    at InEndpoint.epIn.stopPoll (/home/drom/work/lab/drom/ecp5/lib/ftdi.js:99:22)
    at Object.onceWrapper (events.js:273:13)
    at InEndpoint.emit (events.js:182:13)
    at Transfer.transferDone (/home/drom/work/lab/drom/ecp5/node_modules/usb/usb.js:340:10)
```
How come it is still active?
Do I need to clean data buffer after `.on('data', ...)` callback?

kevinmehall, 2018-06-16:
Yeah, I think this is a bug -- it looks like once you `stopPoll`, on an endpoint, you can't `startPoll` again because nothing ever clears the [`pollTransfers`](https://github.com/tessel/node-usb/blob/master/usb.js#L284) property.

That said, I'm not sure what you're trying to do, but you need any kind of control over when transfers are being made, you should probably use `transfer()` to make individual transfers rather than starting and stopping polling. The polling mechanism is just a convenience for keeping a pool of transfers in flight, with a little bit of optimization by reusing the libusb structures.

drom, 2018-06-16:
You are right. I am mostly using `.transfer()` function. That said, I was surprised and puzzled about this behavior of the polling mechanism.

PR #276 (2018-10-28): fix endpoint polling status check
If you call startPoll, stopPoll & startPool on an endpoint you will always get a "Polling already active" exception because the polling state is not properly checked.

This resolves #249
closed 2018-10-28 by kevinmehall