node-usb/node-usb #254: How to force device.close()?

bestplay, 2018-07-28:
My device hangs when sending transfer sometimes.

I need to force `devcie.close()`

but i got `Error: Can't close device with a pending request`

OS: windows 10.
Node: 8.11.3

kevinmehall, 2018-08-20:
You can't close a device until all its transfers are completed or cancelled. `.stopPoll()` will cancel the transfers used in polling, but there is not currently a way to cancel transfers created by the `.transfer()` method. You could use a short timeout, though, and wait for them to time out.

closed 2018-08-20 by kevinmehall

Hexxeh, 2022-08-01:
WebUSB version doesn't have timeouts, how can this be worked around there? I've tried resetting the device but that disconnects other processes using other ports on the same device which is undesirable.

mildsunrise, 2022-08-01:
now that you mention it, WebUSB mandates that a .close() aborts all pending transfers? if we aren't doing that, maybe we should -- especially given that WebUSB gives no way of aborting transfers whatsoever

RReverser, 2023-11-21:
Yup, I ran into this incompatibility between browser and Node.js WebUSB variants as well. I believe this issue should be reopened.