node-usb/node-usb #263: using node-usb with nwjs

chipimix, 2018-08-21:
Hi!

I'm trying to use [this node_module ](https://github.com/sandeepmistry/node-sensortag)with nwjs. My code works when I run it on node.js.
I'm working on a windows10 x64 OS and I've followed [this non-LTS Releases doc](http://docs.nwjs.io/en/latest/For%20Users/Advanced/Use%20Native%20Node%20Modules/):
As such, I run the following commands with no issues:
```
npm install -g nw-gyp  
set PYTHON=C:\Python27\python.exe 
set npm_config_target=0.32.3
set npm_config_arch=x64
set npm_config_runtime=node-webkit
set npm_config_build_from_source=true
set npm_config_node_gyp=C:\Users\DanielRocha\AppData\Roaming\npm\node_modules\nw-gyp\bin\nw-gyp.js
npm install sensortag --msvs_version=2017
```
package.json contents:
```
{
  "name": "helloworld",
  "main": "helloSensortag.html",
  "window": {
    "toolbar": true
  },
  "dependencies": {
    "sensortag": "^1.3.0"
  }
}
```
I run the commandnw.exe . in the correct directory, with no issue.
However, upon page inspection, I get the following error on the console:
```
Uncaught Error: A dynamic link library (DLL) initialization routine failed.
\\?\C:\Users\DanielRocha\Desktop\ibeb\helloNWJS\node_modules\usb\src\binding\usb_bindings.node
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:760:18)
    at Module.load (internal/modules/cjs/loader.js:623:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:562:12)
    at Function.Module._load (internal/modules/cjs/loader.js:554:3)
    at Module.require (internal/modules/cjs/loader.js:661:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (C:\Users\DanielRocha\Desktop\ibeb\helloNWJS\node_modules\usb\usb.js:5:38)
    at Module._compile (internal/modules/cjs/loader.js:713:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:730:10)
    at Module.load (internal/modules/cjs/loader.js:623:32)
```
I have checked the file C:\Users\DanielRocha\Desktop\ibeb\helloNWJS\node_modules\usb\src\binding\usb_bindings.node exists and is human-incomprehensible hex code thingamabob.
🤔
I figure this issue is related with the usb node module.
I have tried to use nw-gyp in the usb directory as I thought the issue could be related with differences in build platforms/versions:

`nw-gyp configure --target=0.32.3`
and
`nw-gyp build --target=0.32.3`

However the `nw-gyp configure --target=0.32.3` resulted in the following error:
```
gyp: Undefined variable module_name in binding.gyp while trying to load binding.gyp
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (C:\Users\DanielRocha\AppData\Roaming\npm\node_modules\nw-gyp\lib\configure.js:359:16)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:239:12)
gyp ERR! System Windows_NT 10.0.17134
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\DanielRocha\\AppData\\Roaming\\npm\\node_modules\\nw-gyp\\bin\\nw-gyp.js" "configure" "---target=0.32.3"
gyp ERR! cwd C:\Users\DanielRocha\Desktop\ibeb\helloNWJS\node_modules\usb
gyp ERR! node -v v10.9.0
gyp ERR! nw-gyp -v v3.6.3
gyp ERR! not ok
```

Does anyone have a clue what am I doing wrong?
Should I just use a LTS version of node.js and respective nw.js?

Thanks in advance!🙇

kevinmehall, 2018-08-21:
This looks like #203 
> You have to build with `npm install --build-from-source` or `./node_modules/.bin/node-pre-gyp -v rebuild`, not node-gyp directly.

But I'm not sure what the equivalent for `nw` is.

chipimix, 2018-08-22:
Thanks for shedding some light @kevinmehall.
Using --build-from-source resulted in the following output:

```
> usb@1.3.3 install C:\Users\DanielRocha\Desktop\ibeb\helloNWJS\node_modules\usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp WARN Using needle for node-pre-gyp https download
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  core.c
  descriptor.c
  hotplug.c
  io.c
  strerror.c
  sync.c
  poll_windows.c
  threads_windows.c
  windows_usb.c
c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\os\windows_usb.c(586): warning C4244: '=': c
onversion from '__int64' to 'unsigned int', possible loss of data [C:\Users\DanielRocha\Desktop\ibeb\helloNWJS\node_mod
ules\usb\build\libusb.vcxproj]
c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\os\windows_usb.c(1423): warning C4244: '=':
conversion from '__int64' to 'unsigned int', possible loss of data [C:\Users\DanielRocha\Desktop\ibeb\helloNWJS\node_mo
dules\usb\build\libusb.vcxproj]
c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\os\windows_usb.c(2558): warning C4018: '<':
signed/unsigned mismatch [C:\Users\DanielRocha\Desktop\ibeb\helloNWJS\node_modules\usb\build\libusb.vcxproj]
  win_delay_load_hook.cc
  libusb.vcxproj -> C:\Users\DanielRocha\Desktop\ibeb\helloNWJS\node_modules\usb\build\Release\\libusb.lib
  node_usb.cc
  device.cc
  transfer.cc
  win_delay_load_hook.cc
c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\libusb.h(740): warning C4200: nonstandard ex
tension used: zero-sized array in struct/union (compiling source file ..\src\node_usb.cc) [C:\Users\DanielRocha\Desktop
\ibeb\helloNWJS\node_modules\usb\build\usb_bindings.vcxproj]
  c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\libusb.h(740): note: This member will be i
  gnored by a defaulted constructor or copy/move assignment operator (compiling source file ..\src\node_usb.cc)
c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\libusb.h(740): warning C4200: nonstandard ex
tension used: zero-sized array in struct/union (compiling source file ..\src\device.cc) [C:\Users\DanielRocha\Desktop\i
beb\helloNWJS\node_modules\usb\build\usb_bindings.vcxproj]
  c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\libusb.h(740): note: This member will be i
  gnored by a defaulted constructor or copy/move assignment operator (compiling source file ..\src\device.cc)
c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\libusb.h(740): warning C4200: nonstandard ex
tension used: zero-sized array in struct/union (compiling source file ..\src\transfer.cc) [C:\Users\DanielRocha\Desktop
\ibeb\helloNWJS\node_modules\usb\build\usb_bindings.vcxproj]
  c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\libusb.h(740): note: This member will be i
  gnored by a defaulted constructor or copy/move assignment operator (compiling source file ..\src\transfer.cc)
c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\libusb.h(771): warning C4200: nonstandard ex
tension used: zero-sized array in struct/union (compiling source file ..\src\node_usb.cc) [C:\Users\DanielRocha\Desktop
\ibeb\helloNWJS\node_modules\usb\build\usb_bindings.vcxproj]
  c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\libusb.h(771): note: This member will be i
  gnored by a defaulted constructor or copy/move assignment operator (compiling source file ..\src\node_usb.cc)
c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\libusb.h(771): warning C4200: nonstandard ex
tension used: zero-sized array in struct/union (compiling source file ..\src\device.cc) [C:\Users\DanielRocha\Desktop\i
beb\helloNWJS\node_modules\usb\build\usb_bindings.vcxproj]
  c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\libusb.h(771): note: This member will be i
  gnored by a defaulted constructor or copy/move assignment operator (compiling source file ..\src\device.cc)
c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\libusb.h(771): warning C4200: nonstandard ex
tension used: zero-sized array in struct/union (compiling source file ..\src\transfer.cc) [C:\Users\DanielRocha\Desktop
\ibeb\helloNWJS\node_modules\usb\build\usb_bindings.vcxproj]
  c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\libusb.h(771): note: This member will be i
  gnored by a defaulted constructor or copy/move assignment operator (compiling source file ..\src\transfer.cc)
c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\libusb.h(1265): warning C4200: nonstandard e
xtension used: zero-sized array in struct/union (compiling source file ..\src\node_usb.cc) [C:\Users\DanielRocha\Deskto
p\ibeb\helloNWJS\node_modules\usb\build\usb_bindings.vcxproj]
  c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\libusb.h(1265): note: This member will be
  ignored by a defaulted constructor or copy/move assignment operator (compiling source file ..\src\node_usb.cc)
c:\users\danielrocha\desktop\ibeb\hellonwjs\node_modules\usb\libusb\libusb\libusb.h(1265): warning C4200: nonstandard e
xtension used: zero-sized array in struct/union (compiling source file ..\src\device.cc) [C:\Users\DanielRocha\Desktop\
[... cut ...]
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\DanielRocha\\Desktop\\ibeb\\helloNWJS\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "-v" "--runtime=node-webkit" "--target=0.32.3" "rebuild"
node-pre-gyp ERR! cwd C:\Users\DanielRocha\Desktop\ibeb\helloNWJS
node-pre-gyp ERR! node -v v10.9.0
node-pre-gyp ERR! node-pre-gyp -v v0.10.3
node-pre-gyp ERR! not ok
```
I tried removing /Users/username/.node-gyp but it didn't fix this either.
Has anyone succesfully used node-usb with nwjs?
Thanks again!

kevinmehall, 2018-08-23:
First one indeed looks like it built successfully.

For the second one, you would run that in `node_modules/usb`; no need to add that to your own package.json.

chipimix, 2018-08-24:
For the 1st one it built successfully but still has the same DLL error upon frontend page inspection ☹️ 
For the second one:
I ran it in node_modules/usb as u suggested: 

`node-pre-gyp -v --runtime=node-webkit --target=0.32.3 rebuild`

which resulted in the following error:
```
node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ 'C:\\Program Files\\nodejs\\node.exe',
node-pre-gyp verb cli   'C:\\Users\\DanielRocha\\AppData\\Roaming\\npm\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp',
node-pre-gyp verb cli   '-v',
node-pre-gyp verb cli   '--runtime=node-webkit',
node-pre-gyp verb cli   '--target=0.32.3',
node-pre-gyp verb cli   'rebuild' ]
node-pre-gyp info using node-pre-gyp@0.11.0
node-pre-gyp info using node@10.9.0 | win32 | x64
node-pre-gyp verb command rebuild []
node-pre-gyp verb command clean []
[usb] Removing "C:\Users\DanielRocha\Desktop\ibeb\helloNWJS\node_modules\usb\src\binding"
node-pre-gyp verb command build [ 'rebuild' ]
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp.cmd clean' (Error: spawn node-gyp.cmd ENOENT)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\DanielRocha\AppData\Roaming\npm\node_modules\node-pre-gyp\lib\util\compile.js:77:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:237:12)
node-pre-gyp ERR! stack     at onErrorNT (internal/child_process.js:406:16)
node-pre-gyp ERR! stack     at process._tickCallback (internal/process/next_tick.js:63:19)
node-pre-gyp ERR! System Windows_NT 10.0.17134
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\DanielRocha\\AppData\\Roaming\\npm\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "-v" "--runtime=node-webkit" "--target=0.32.3" "rebuild"
node-pre-gyp ERR! cwd C:\Users\DanielRocha\Desktop\ibeb\helloNWJS\node_modules\usb
node-pre-gyp ERR! node -v v10.9.0
node-pre-gyp ERR! node-pre-gyp -v v0.11.0
node-pre-gyp ERR! 
```

Alternatively, I tried Electron and - coherently - I got a similar error message upon page inspection:
```
Uncaught Error: A dynamic link library (DLL) initialization routine failed.
\\?\C:\Users\DanielRocha\Desktop\electronSensorTag1\node_modules\usb\src\binding\usb_bindings.node
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:172:20)
    at Object.Module._extensions..node (module.js:671:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:172:20)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (module.js:586:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (C:\Users\DanielRocha\Desktop\electronSensorTag1\node_modules\usb\usb.js:5:38)
    at Object.<anonymous> (C:\Users\DanielRocha\Desktop\electronSensorTag1\node_modules\usb\usb.js:417:3)
```

chipimix, 2018-08-24:
I ran `./node_modules/.bin/electron-rebuild.cmd` [as advised here](https://github.com/electron/electron/blob/v0.37.2/docs/tutorial/using-native-node-modules.md#using-native-node-modules) in my project directory and now the DLL error is gone!
Thanks for the tips @kevinmehall! 😃

closed 2018-08-24 by chipimix