node-usb/node-usb #264: usb in electron error: usb_bindings.node is not a valid Win32 applicatio

kingov, 2018-08-25:
electron is 2.0.8
platform is macOS 10.13
node is 8.9.4

when i pack windows and run,  an  error occurred .   but pack mac and run it's ok, 

[image]

i try this, but not help
`
npm install --save-dev electron-rebuild
  ./node_modules/.bin/electron-rebuild
  .\node_modules\.bin\electron-rebuild.cmd
`

kevinmehall, 2018-08-25:
Does this help? Haven't tried it myself but it's been reported by others to fix this.

https://github.com/electron/electron/blob/v0.37.2/docs/tutorial/using-native-node-modules.md#using-native-node-modules

DalderupMaurice, 2019-03-16:
This issue still occurs on electron 2.0.18 when including @ledgerhq/hw-transport-node-hid (which has a dependency on usb@1.5.0)

I was already building to native dependencies and tried both electron-builder and electron-rebuild.
The error translates back to this library:
[image]

tayvano, 2019-03-29:
Samesies for MyCrypto electron with ledger / USB 1.5.0

We build on mac but USB isn't rebuilding (like node-hid and scrypt do) so it's broken on Windows builds. 

I think this will be resolved with the new release that using prebuilds but it's not ready.

kevinmehall, 2019-05-19:
Closing all old Electron build related issues now that we have prebuilds for Electron in 1.6.0.

closed 2019-05-19 by kevinmehall