node-usb/node-usb #267: i meet this problem,can you tell me how to solve

yexianli, 2018-09-07:
package.json does not exist at F:\package.json
    at Object../node_modules/usb/node_modules/node-pre-gyp/lib/pre-binding.js.exports.find (pre-binding.js:18)

gui Unhandled Error: Error: package.json does not exist at F:\package.json
    at Object../node_modules/usb/node_modules/node-pre-gyp/lib/pre-binding.js.exports.find (http://localhost:8601/lib.min.js:529305:15)
    at Object.<anonymous> (http://localhost:8601/lib.min.js:559081:27)
    at Object../node_modules/usb/usb.js (http://localhost:8601/lib.min.js:559495:30)
    at __webpack_require__ (http://localhost:8601/lib.min.js:79:30)
    at GUIComponent.spyUsb (http://localhost:8601/lib.min.js:574139:23)
    at GUIComponent.componentDidMount (http://localhost:8601/lib.min.js:574154:18)
    at commitLifeCycles (http://localhost:8601/lib.min.js:345362:24)
    at commitAllLifeCycles (http://localhost:8601/lib.min.js:346538:9)
    at HTMLUnknownElement.callCallback (http://localhost:8601/lib.min.js:337134:14)
    at Object.invokeGuardedCallbackDev (http://localhost:8601/lib.min.js:337173:16)
Component stack: 
    in GUIComponent (created by Connect(GUIComponent))
    in Connect(GUIComponent) (created by InjectIntl(Connect(GUIComponent)))
    in InjectIntl(Connect(GUIComponent)) (created by GUI)
    in GUI (created by Connect(GUI))
    in Connect(GUI) (created by VMListener)
    in VMListener (created by Connect(VMListener))
    in Connect(VMListener) (created by ProjectLoaderComponent)
    in ProjectLoaderComponent (created by ErrorBoundaryWrapper)
    in ErrorBoundary (created by ErrorBoundaryWrapper)
    in ErrorBoundaryWrapper (created by AppStateWrapper)
    in IntlProvider (created by Connect(IntlProvider))
    in Connect(IntlProvider) (created by AppStateWrapper)
    in Provider (created by AppStateWrapper)
    in AppStateWrapper (created by HashParserComponent)
    in HashParserComponent

kevinmehall, 2018-09-07:
Are you trying to use node-usb in a React app built with webpack? It won't work in a browser.

closed 2018-09-07 by kevinmehall

yexianli, 2018-09-07:
i haven't let it work browse, i let it work electron+react

kevinmehall, 2018-09-08:
See https://github.com/electron/electron/blob/master/docs/tutorial/using-native-node-modules.md

yexianli, 2018-09-08:
thank you very much