node-usb/node-usb #272: Not found usb_bindings-v1.3.3-electron-v2.0-win32-x64.tar.gz

nikas64527, 2018-10-05:
Hello! Please, help:
When my project building here is an error: 
`node-pre-gyp WARN Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.3.3/usb_bindings-v1.3.3-electron-v2.0-win32-x64.tar.gz 
`
Then node-gyp try to compile it himself (its very long - about 270sec and it failed in 70%).

How can I fix it?

fangjunai, 2018-10-23:
Error output:
node-pre-gyp WARN Using request for node-pre-gyp https download
node-pre-gyp WARN Using request for node-pre-gyp https download
node-pre-gyp WARN Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.3.3/usb_bindings-v1.3.3-electron-v2.0-win32-x64.tar.gz
node-pre-gyp WARN Pre-built binaries not found for usb@1.3.3 and electron@2.0.9 (electron-v2.0 ABI, unknown) (falling back to source compile with node-gyp)
node-pre-gyp WARN Pre-built binaries not installable for usb@1.3.3 and electron@2.0.9 (electron-v2.0 ABI, unknown) (falling back to source compile with node-gyp)
node-pre-gyp WARN Hit error Connection closed while downloading tarball file
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:240:12)
gyp ERR! System Windows_NT 10.0.17763
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--module=D:\\dev\\Electron\\vue2\\node_modules\\usb\\src\\binding\\usb_bindings.node" "--module_name=usb_bindings" "--module_path=D:\\dev\\Electron\\vue2\\node_modules\\usb\\src\\binding" "--napi_version=3" "--node_abi_napi=napi"
gyp ERR! cwd D:\dev\Electron\vue2\node_modules\usb
gyp ERR! node -v v10.12.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok

kevinmehall, 2018-10-23:
https://github.com/electron/electron/blob/master/docs/tutorial/using-native-node-modules.md#using-native-node-modules

closed 2018-10-23 by kevinmehall