node-usb/node-usb #277: Interface.claim fails on some macos devices

reed-jones, 2018-11-01:
I have an electron app which uses node-usb. When I build my project on a 2015 MacBook Pro, It works fine on my device, and all other sub-2015 devices I tried, but crashes with `LIBUSB_ERROR_OTHER` when I use`Interface.claim` (I get errno: -99) on newer (2017) MacBooks/iMacs. When I build it on a 2017 iMac, it works on all tested devices. I'm not sure the error is tied to the year, but that's as far as I got debugging. I'm rebuilding node-usb using electron-rebuild.
Any help would be appreciated.

tzvetelin-vassilev, 2018-11-07:
I have experience the same problem. When app is built under Mojave LIBUSB_ERROR_OTHER is thrown under High Sierra, when Interface.claim is executed.

    at Interface.claim (/Users/../app/Contents/Resources/app/node_modules/usb/usb.js:187)
 
Under Mojave app is working properly.

reed-jones, 2018-11-07:
For me, when the app is built on Mojave (2015 MacBook), and the error throws on both High Sierra, and Mojave 2017 computers.

reed-jones, 2019-02-11:
I can no long reproduce this, so I have closed this issue.

closed 2019-02-11 by reed-jones

KeepKeyBrett, 2019-02-18:
I have produced this error as well. I'm using a library, `webusb` which is dependent on `usb`. When I build my electron app for production and try to claim the interface, I get `LIBUSB_ERROR_OTHER` for any machine not running Mojave.

fridgerator, 2019-03-12:
Also having this issue.  Works fine on Mojave, however in High Sierra I am unable to cliam interface `0`, it fails with `LIBUS_ERROR_OTHER`.

fridgerator, 2019-03-12:
Just tested on a different High Sierra computer and I can claim the interface with no issues....  🤕

tzvetelin-vassilev, 2019-03-14:
It is working under High Sierra when app is built on High Sierra, but when app is build on Mojave and installed on High Sierra, then not working any more.