node-usb/node-usb #282: why I use electron-rebuild to rebuild usb then run my project and find the error?

yexianli, 2018-11-09:
pre-binding.js:18 Uncaught Error: package.json does not exist at D:\package.json
    at Object../node_modules/usb/node_modules/node-pre-gyp/lib/pre-binding.js.exports.find (pre-binding.js:18)
    at Object.<anonymous> (usb.js:3)
    at Object../node_modules/usb/usb.js (usb.js:415)
    at __webpack_require__ (bootstrap:78)
    at Object../src/containers/blocks.jsx (blocks.jsx:27)
    at __webpack_require__ (bootstrap:78)
    at Object../src/components/gui/gui.jsx (gui.jsx:12)
    at __webpack_require__ (bootstrap:78)
    at Object../src/containers/gui.jsx (gui.jsx:25)
    at __webpack_require__ (bootstrap:78)

tzvetelin-vassilev, 2018-11-12:
You should exclude native libraries from webpack. You need externals option in your webpack config.

kevinmehall, 2019-05-19:
Closing all old Electron build related issues now that we have prebuilds for Electron in 1.6.0.

closed 2019-05-19 by kevinmehall

PaulBernier, 2019-06-18:
Hi @kevinmehall could hint how to properly leverage your new prebuilds for Electron? My problem is that when I run `npm install` it will fetch the prebuild for the node.js version I am using in my terminal and not the Node.js of Electron. Is there a way (command) to force to fetch specifically the Electron prebuild? A line in the README of the project would be super useful! Thanks.

kevinmehall, 2019-06-19:
@PaulBernier I haven't personally used Electron. [prebuild-install](https://github.com/prebuild/prebuild-install) mentions `--runtime` and `--target` that look relevant. If you find a good way, please submit a PR to the README.

kevinmehall, 2019-06-19:
`electron-rebuild` might be what you're looking for? https://github.com/electron/electron-rebuild#how-can-i-integrate-this-with-prebuild