node-usb/node-usb #284: Android support via Cordova nodejs-mobile plugin

Aicos, 2018-11-15:
I'm attempting to use node-usb on android, but building (cross compiling?) from osx.  I understand that android/arm versions of linux_udev.o (libudev_dev) are required. 

The build process which is handled by Cordova 7 (and  the nodejs-mobile plugin) seems try and compile linux_udev.o but fails.  I have no idea where to aquire this lib for android/arm or if node-usb even supports android.

Are there any solutions for this?

Thanks.

```
$ cordova run android

ANDROID_HOME=/Users/aicos/android/
JAVA_HOME=/Library/Java/JavaVirtualMachines/jdk1.8.0_192.jdk/Contents/Home
Subproject Path: CordovaLib
/Users/aicos/.nvm/versions/node/v8.2.1/bin/npm
The Task.leftShift(Closure) method has been deprecated and is scheduled to be removed in Gradle 5.0. Please use Task.doLast(Action) instead.
	at build_dgnimap2xmuwmli9y3r3di9to.run(/Users/aicos/proj/updtr/platforms/android/build.gradle:142)
The JavaCompile.setDependencyCacheDir() method has been deprecated and is scheduled to be removed in Gradle 4.0.
Incremental java compilation is an incubating feature.
The TaskInputs.source(Object) method has been deprecated and is scheduled to be removed in Gradle 4.0. Please use TaskInputs.file(Object).skipWhenEmpty() instead.
:CopyNodeProjectAssetsarmeabi-v7a UP-TO-DATE
:MakeToolchainarmeabi-v7a UP-TO-DATE
:BuildNpmModulesarmeabi-v7anpm info it worked if it ends with ok
npm verb cli [ '/Users/aicos/.nvm/versions/node/v8.2.1/bin/node',
npm verb cli   '/Users/aicos/.nvm/versions/node/v8.2.1/bin/npm',
npm verb cli   '--verbose',
npm verb cli   'rebuild',
npm verb cli   '--build-from-source' ]
npm info using npm@5.3.0
npm info using node@v8.2.1
npm info readInstalled object
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project',
npm verb rebuild   'nodejs-mobile-sample-project@0.1.0' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/usb',
npm verb rebuild   'usb@1.5.0' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/nan',
npm verb rebuild   'nan@2.11.1' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/node-pre-gyp',
npm verb rebuild   'node-pre-gyp@0.11.0' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/detect-libc',
npm verb rebuild   'detect-libc@1.0.3' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/mkdirp',
npm verb rebuild   'mkdirp@0.5.1' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/minimist',
npm verb rebuild   'minimist@0.0.8' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/needle',
npm verb rebuild   'needle@2.2.4' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/debug',
npm verb rebuild   'debug@2.6.9' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/ms',
npm verb rebuild   'ms@2.0.0' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/iconv-lite',
npm verb rebuild   'iconv-lite@0.4.24' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/safer-buffer',
npm verb rebuild   'safer-buffer@2.1.2' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/sax',
npm verb rebuild   'sax@1.2.4' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/nopt',
npm verb rebuild   'nopt@4.0.1' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/abbrev',
npm verb rebuild   'abbrev@1.1.1' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/osenv',
npm verb rebuild   'osenv@0.1.5' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/os-homedir',
npm verb rebuild   'os-homedir@1.0.2' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/os-tmpdir',
npm verb rebuild   'os-tmpdir@1.0.2' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/npm-packlist',
npm verb rebuild   'npm-packlist@1.1.12' ]
npm verb rebuild path, id [ '/Users/aicos/proj/updtr/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/ignore-walk',
[... 264 lines cut ...]
npm ERR! errno 1
npm ERR! usb@1.5.0 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb@1.5.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm verb exit [ 1, true ]

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/aicos/.npm/_logs/2018-11-15T14_00_52_564Z-debug.log
:BuildNpmModulesarmeabi-v7a FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':BuildNpmModulesarmeabi-v7a'.
> Process 'command 'npm'' finished with non-zero exit value 1

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 10.56 secs
```

kevinmehall, 2018-11-15:
udev is not present on Android as far as I know, so this library will not work unmodified. Your best path forward is probably to use libusb with [this patch](https://github.com/libusb/libusb/pull/242), and add a node-usb wrapper for the `libusb_wrap_fd` API it adds.

thegecko, 2023-02-26:
We have native android builds now, please create a new issue if there are still problems.

closed 2023-02-26 by thegecko