node-usb/node-usb #290: Remove `aws-sdk` dev dependency

HipsterBrown, 2018-12-19:
This dependency has been causing issues (#215) and does not appear to be used anymore since `node-usb` switched to using `node-pre-gyp-github` to host binaries on GitHub a few years ago. 

I think we can safely remove it now.

milonite, 2018-12-27:
Agree with this #https://github.com/LedgerHQ/ledgerjs/issues/242

HipsterBrown, 2019-01-03:
Ok this is open to anyone as a great first contribution. Otherwise, I'll add it to my Bug Fix Friday list. 😄

rohan-malik, 2019-02-02:
Could you kindly assign this bug to me?

rohan-malik, 2019-02-04:
I've added a pull request for this issue:

https://github.com/tessel/node-usb/pull/295/commits/fce72ce7664e9e6e5753fdbc684a012a6e710386

jrobeson, 2019-03-19:
Fixed in https://github.com/tessel/node-usb/pull/300

closed 2019-03-19 by kevinmehall