node-usb/node-usb #301: Interface.claim() still failing on certain Mac OSX devices

adamtLICOR, 2019-03-12:
I'm facing the exact same crash referenced in this [Issue](https://github.com/tessel/node-usb/issues/277).

To sum it up, after any attempt to use `Interface.claim()`, I get a crash with a **LIBUSB_ERROR_OTHER** error.

I am using a MacBook Pro (Retina, 15-inch, Mid 2015) running High Sierra.

Interestingly enough, my colleague is using the exact same model of MacBook, but is running Mojave -- he does not have this problem.

Additionally, both our office iMac (running High Sierra) and a High Sierra VM (running on my co-worker's Mojave MacBook) don't have this problem.

I noticed the Issue linked above is officially closed -- is the official stance just "upgrade to Mojave already, dude," or is there another known work-around for this problem?

gniezen, 2019-03-22:
The issue you linked to was closed by the OP because they couldn't reproduce it anymore for some reason.

The actual issue is in libusb and described here: https://github.com/libusb/libusb/issues/519

TL;DR: IOUSBLib.h is broken and Apple would need to update XCode to fix it. In the meanwhile, a similar fix like https://github.com/libusb/libusb/commit/efa1b280e66c0a62d892bddf826e624362d1e50a could be used?

adamtLICOR, 2019-03-22:
@gniezen Okay, great!

I'll look into that -- thank you so much!

thegecko, 2021-11-08:
Closing due to inactivity, please reopen this issue if it persists with the latest release.

closed 2021-11-08 by thegecko