node-usb/node-usb #302: LIBUSB_ERROR_ACCESS

jfdelarosa, 2019-03-14:
I know this is a known issue, but all the post are kind of old.
Is there already a solution for LIBUSB_ERROR_ACCESS on mac?

Thanks

zhyjor, 2019-08-09:
It seems that i have a same problem:

tools/node_modules/usb/usb.js:280
	this.device.__claimInterface(this.id)
	            ^

Error: LIBUSB_ERROR_ACCESS
    at Interface.claim (/Users/yjzh/demo/mtp-demo/node-tools/node_modules/usb/usb.js:280:14)
    at Object.<anonymous> (/Users/yjzh/demo/mtp-demo/node-tools/usb.js:29:13)
    at Module._compile (internal/modules/cjs/loader.js:721:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:732:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:560:12)
    at Function.Module._load (internal/modules/cjs/loader.js:552:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:774:12)
    at executeUserCode (internal/bootstrap/node.js:342:17)
    at startExecution (internal/bootstrap/node.js:276:5)

markosole, 2020-04-22:
It's same here on windows. Linux works, but but you have to give access to devices. I did not try it on Mac yet, I need it for windows and all I am doing now is trying to make it work on win 10.

Try running script with sudo, that should help.

koalazak, 2020-04-27:
Same `LIBUSB_ERROR_ACCESS` on mac `10.15.1` with this simple code:

```javacript
const usb = require('usb')

const device = usb.findByIds(0xFF, 0xFF); // edited for posting
device.open();
// connected OK
const interface = device.interface(2);
console.log(interface.isKernelDriverActive()); // print false;

interface.claim(); // With this line I got the LIBUSB_ERROR_ACCESS error

const endpoint = interface.endpoint(0x84);

console.log(endpoint); //well selected

endpoint.transfer(64, function callback(error, data) {
  console.log(error); // got LIBUSB_ERROR_NOT_FOUND error
  console.log(data);
})
```

im missing something? thank you

markosole, 2020-04-27:
If you can't open device with `interface.claim();` than you do not have permissions to take over control over your device. Try running your script with `sudo`.
I do not have enough experience with mac and can't tell you is going to work if you edit some files for your USB devices (like linux has). 
I've red something about libusb drivers for Mac but as I said, I've no experience with it but more than using finder and Xcode.

kevinmehall, 2020-04-27:
On macOS, regular users have permission to access USB devices by default. LIBUSB_ERROR_ACCESS happens most commonly there is already a kernel driver loaded for the device. There is not a great way to detach the kernel driver on macOS -- the `isKernelDriverActive` / `detachKernelDriver` APIs only work on Linux: https://github.com/libusb/libusb/wiki/FAQ#how-can-i-run-libusb-applications-under-mac-os-x-if-there-is-already-a-kernel-extension-installed-for-the-device

It is not advisable to run apps with `sudo` unnecessarily on Linux either. Set up udev rules to give your user account access to the device.

koalazak, 2020-04-28:
Regarding that FAQ, unload the driver is not possible for HID decices. `Take note this may not be possible for drivers like USB HID since it may be used by other USB HID device.`
And my IN endpoint is in a HID interface of the device :/
Now I see on this FAQ https://github.com/tessel/node-usb/issues/182 that this problem was already reported.
Is there any way to get data from a HID IN endpoint with this library? or I should try with node-hid?

thank you!

kevinmehall, 2020-04-28:
You should use node-hid (libhid) for a HID device. That uses the kernel's HID driver instead of trying to replace it.

koalazak, 2020-04-28:
Ok, thank you Kevin! now my problem is that node-hid seems a very low level thing...I have no idea what I should "write()" to the device to get data on the `data` event. Do you know the format of that data? 
This is the packet that Im trying to emulate to get data back: (sniffed with wireshark)
```
Frame 5454: 27 bytes on wire (216 bits), 27 bytes captured (216 bits) on interface 
USB URB
    [Source: host]
    [Destination: 1.7.4]
    USBPcap pseudoheader length: 27
    IRP ID: 0xffffd285cba7b8a0
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER (0x0009)
    IRP information: 0x00, Direction: FDO -> PDO
    URB bus id: 1
    Device address: 7
    Endpoint: 0x84, Direction: IN
        1... .... = Direction: IN (1)
        .... 0100 = Endpoint number: 4
    URB transfer type: URB_INTERRUPT (0x01)
    Packet Data Length: 0
    [Response in: 5461]
    [bInterfaceClass: HID (0x03)]
```
```
0000   1b 00 a0 b8 a7 cb 85 d2 ff ff 00 00 00 00 09 00   ................
0010   00 01 00 07 00 84 01 00 00 00 00                  ...........
```

thank you in advance

kevinmehall, 2020-04-28:
It's higher-level than node-usb since it deals with reports instead of endpoints and transfers. If your device uses HID "correctly" the report descriptors will tell you the format and meaning of the HID reports in both directions. (Google "HID Report Descriptor"). Many "HID" devices don't usefully implement HID though, but just use it to pass arbitrary data because it simplified driver install on old versions of Windows. In that case, you'll have to reverse-engineer what the data means.

daniellizik, 2021-12-08:
I get this error but only after printing 64 times. On the 65th attempt `interface.claim()` throws the libusb access error

markosole, 2021-12-08:
Just an idea from the top of my head, try restarting the sequence and device after 64th read (command to restart or re-init) and see what's happening after.
It could be that device does not "clear" the buffer. I am not 100% sure about this but try.

daniellizik, 2021-12-08:
@markosole might be a driver level issue https://github.com/libusb/libusb/issues/16

daniellizik, 2021-12-09:
@markosole I got around this issue by doing the printing in a forked thread. When the thread reaches its specified max print amount (64 or 32 depending on the os architecture) the parent thread spawns a new one and kills the current thread.

markosole, 2021-12-09:
@daniellizik  that's great workaround. Good to know, I've tested with two different devices and in a limited environment.

esinanturan, 2021-12-11:
> @markosole I got around this issue by doing the printing in a forked thread. When the thread reaches its specified max print amount (64 or 32 depending on the os architecture) the parent thread spawns a new one and kills the current thread.

Can you provide an example , I am having this issue with windows in electron and I don't know what you exactly doing.

thegecko, 2023-02-26:
I think this issue is a usage problem, the OS may block device access and this error is being thrown by libusb. Perhaps workarounds should be documented as part of #182

closed 2023-02-26 by thegecko

isramv, 2024-05-31:
# I am on Ubuntu for me this was the fix... generated with GTP:

To set up `udev` rules to give your user account access to a USB device on Linux, you can follow these steps:

### Step 1: Identify the USB Device

1. Plug in the USB device.
2. Open a terminal and list all connected USB devices:
   ```bash
   lsusb
   ```
3. Note the vendor ID and product ID of the USB device. The output will look something like this:
   ```
   Bus 001 Device 002: ID 1234:5678 Sample USB Device
   ```
   Here, `1234` is the vendor ID and `5678` is the product ID.

### Step 2: Create a `udev` Rule

1. Create a new `udev` rule file in the `/etc/udev/rules.d/` directory. You can name it something like `99-usb.rules`:
   ```bash
   sudo nano /etc/udev/rules.d/99-usb.rules
   ```
2. Add the following line to the file, replacing `1234` with your vendor ID and `5678` with your product ID:
   ```plaintext
   SUBSYSTEM=="usb", ATTR{idVendor}=="1234", ATTR{idProduct}=="5678", MODE="0660", GROUP="plugdev"
   ```
   This rule sets the permissions of the device to `0660` and assigns it to the `plugdev` group.

### Step 3: Add Your User to the `plugdev` Group

1. Add your user to the `plugdev` group:
   ```bash
   sudo usermod -aG plugdev $USER
   ```
2. Log out and log back in to apply the group changes.

### Step 4: Reload `udev` Rules

1. Reload the `udev` rules to apply the changes:
   ```bash
   sudo udevadm control --reload-rules
   sudo udevadm trigger
   ```

### Step 5: Verify Permissions

1. Unplug and replug the USB device.
2. Check the device permissions to ensure they have been set correctly:
   ```bash
   ls -l /dev/bus/usb/001/002
   ```
   The exact path may vary; adjust it according to your system.

After completing these steps, your user should have access to the USB device without needing `sudo`. If you still encounter issues, make sure that no other kernel drivers are conflicting with the device access.