node-usb/node-usb #305: Cannot build usb module on electron

kentng201, 2019-03-25:
I need to build node-usb module in order to use in electron

After running dev in electon with command "node .electron-vue/dev-runner.js", it showing First Error in below 
And npm install with command "electron-builder install-app-deps", it show Second Error in below, I think it cannot found the tar achieved in the repo? 
* **First Error:** The module '\\?\D:\projects\myproject\node_modules\@serialport\bindings\build\Release\bindings.node'
  was compiled against a different Node.js version using
  NODE_MODULE_VERSION 57. This version of Node.js requires
  NODE_MODULE_VERSION 69. Please try re-compiling or re-installing
  the module (for instance, using `npm rebuild` or `npm install`).
      at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:160:31)
      at Object.Module._extensions..node (internal/modules/cjs/loader.js:722:18)
      at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:160:31)
      at Module.load (internal/modules/cjs/loader.js:602:32)
      at tryModuleLoad (internal/modules/cjs/loader.js:541:12)
      at Function.Module._load (internal/modules/cjs/loader.js:533:3)
      at Module.require (internal/modules/cjs/loader.js:640:17)
      at require (internal/modules/cjs/helpers.js:20:18)
      at bindings (D:\projects\projects\myproject\node_modules\bindings\bindings.js:112:48)
      at Object.<anonymous> (D:\projects\myproject\node_modules\@serialport\bindings\lib\win32.js:1:191)

* **Second Error:** Error: C:\Program Files\nodejs\node.exe exited with code 1
Output:

> usb@1.5.0 install D:\projects\myproject\node_modules\usb
> node-pre-gyp install --fallback-to-build

Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
D:\projects\electron\feedme-restaurant-desktop\node_modules\usb\build\binding.sln.metaproj(2,10): error MSB4132: The tools version "4.0" is unrecognized. Available tools versions are "12.0", "14.0", "2.0", "3.5".
Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\Admin\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=D:\projects\electron\feedme-restaurant-desktop\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=D:\projects\electron\feedme-restaurant-desktop\node_modules\usb\src\binding --napi_version=3 --node_abi_napi=napi --napi_build_version=0 --node_napi_label=electron-v4.1' (1)

Error output:
node-pre-gyp WARN Using request for node-pre-gyp https download 
node-pre-gyp WARN Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.5.0/usb_bindings-v1.5.0-electron-v4.1-win32-x64.tar.gz 
node-pre-gyp WARN Pre-built binaries not found for usb@1.5.0 and electron@4.1.0 (electron-v4.1 ABI, unknown) (falling back to source compile with node-gyp) 
gyp ERR! build error 
gyp ERR! stack Error: `msbuild` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\Admin\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\build.js:262:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Windows_NT 10.0.17134
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Admin\\AppData\\Roaming\\npm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--module=D:\\projects\\electron\\feedme-restaurant-desktop\\node_modules\\usb\\src\\binding\\usb_bindings.node" "--module_name=usb_bindings" "--module_path=D:\\projects\\electron\\feedme-restaurant-desktop\\node_modules\\usb\\src\\binding"
"--napi_version=3" "--node_abi_napi=napi" "--napi_build_version=0" "--node_napi_label=electron-v4.1"
gyp ERR! cwd D:\projects\electron\feedme-restaurant-desktop\node_modules\usb
gyp ERR! node -v v8.11.3
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\Admin\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=D:\projects\electron\feedme-restaurant-desktop\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=D:\projects\electron\feedme-restaurant-desktop\node_modules\usb\src\binding --napi_version=3 --node_abi_napi=napi --napi_build_version=0 --node_napi_label=electron-v4.1' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (D:\projects\electron\feedme-restaurant-desktop\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:126:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:925:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
node-pre-gyp ERR! System Windows_NT 10.0.17134
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "D:\\projects\\electron\\feedme-restaurant-desktop\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd D:\projects\electron\feedme-restaurant-desktop\node_modules\usb
node-pre-gyp ERR! node -v v8.11.3
node-pre-gyp ERR! node-pre-gyp -v v0.11.0
node-pre-gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb@1.5.0 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb@1.5.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Admin\AppData\Roaming\npm-cache\_logs\2019-03-25T03_50_43_222Z-debug.log

    at ChildProcess.childProcess.once.code (D:\projects\electron\feedme-restaurant-desktop\node_modules\builder-util\src\util.ts:241:14)
    at Object.onceWrapper (events.js:317:30)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at maybeClose (internal/child_process.js:925:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
From previous event:
    at spawn (D:\projects\electron\feedme-restaurant-desktop\node_modules\builder-util\src\util.ts:200:3)
    at D:\projects\electron\feedme-restaurant-desktop\node_modules\electron-builder-lib\src\util\yarn.ts:171:5
From previous event:
    at rebuild (D:\projects\electron\feedme-restaurant-desktop\node_modules\electron-builder-lib\out\util\yarn.js:93:22)
    at D:\projects\electron\feedme-restaurant-desktop\node_modules\electron-builder-lib\src\util\yarn.ts:20:11
    at Generator.next (<anonymous>)
    at runCallback (timers.js:810:20)
    at tryOnImmediate (timers.js:768:5)
    at processImmediate [as _immediateCallback] (timers.js:745:5)
From previous event:
    at installOrRebuild (D:\projects\electron\feedme-restaurant-desktop\node_modules\electron-builder-lib\out\util\yarn.js:31:21)
    at D:\projects\electron\feedme-restaurant-desktop\node_modules\electron-builder\src\cli\install-app-deps.ts:58:42
    at Generator.next (<anonymous>)
From previous event:
    at installAppDeps (D:\projects\electron\feedme-restaurant-desktop\node_modules\electron-builder\out\cli\install-app-deps.js:51:21)
    at then (D:\projects\electron\feedme-restaurant-desktop\node_modules\electron-builder\src\cli\cli.ts:48:33)
    at runCallback (timers.js:810:20)
    at tryOnImmediate (timers.js:768:5)
    at processImmediate [as _immediateCallback] (timers.js:745:5)
From previous event:
    at Object.args [as handler] (D:\projects\electron\feedme-restaurant-desktop\node_modules\electron-builder\src\cli\cli.ts:48:33)
    at Object.runCommand (D:\projects\electron\feedme-restaurant-desktop\node_modules\electron-builder\node_modules\yargs\lib\command.js:235:44)
    at Object.parseArgs [as _parseArgs] (D:\projects\electron\feedme-restaurant-desktop\node_modules\electron-builder\node_modules\yargs\yargs.js:1046:30)
    at Object.get [as argv] (D:\projects\electron\feedme-restaurant-desktop\node_modules\electron-builder\node_modules\yargs\yargs.js:989:21)
    at Object.<anonymous> (D:\projects\electron\feedme-restaurant-desktop\node_modules\electron-builder\src\cli\cli.ts:42:15)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:191:16)
    at bootstrap_node.js:612:3

kentng201, 2019-04-05:
Found problem, it is due to I store too much of environment path in Windows

closed 2019-04-05 by kentng201

jocodev1, 2019-05-29:
@kentng201 Can you explain what you actually did to fix your first error?

kentng201, 2019-05-30:
> @kentng201 Can you explain what you actually did to fix your first error?

It is related to the second error, due to my windows env path is full. 
After cleanup some unused path, I install windows build tools 2015, and rebuild dependencies "node_modules/.bin/electron-builder install-app-deps", it will work again