node-usb/node-usb #307: Claim always retuns busy

gberth, 2019-04-07:
Stuck (with incompetence)

Simple program, and debug attached, tried with different usb devices, same result. Running on Ubuntu 18.04
Kode:

var usb = require('usb')

var term = usb.findByIds(1659, 8963)
usb.setDebugLevel(4)

term.open()

var endpoints = term.interfaces[0].endpoints

var inEndpoint = endpoints[0]

var outEndpoint = endpoints[1]

// term.claim()
// console.log(endpoints)
inEndpoint.transferType = 2
// console.log(inEndpoint)
term.interfaces[0].claim()

inEndpoint.startPoll(1, 8)

inEndpoint.on('data', function (data) {
  console.log('**********')
  console.log(data)
})
inEndpoint.on('error', function (error) {
  console.log('on error', error)
})

inEndpoint.on('end', function () {
  console.log('on end')
})
outEndpoint.on('data', function (data) {
  console.log(data)
})

outEndpoint.on('error', function (error) {
  console.log(error)
})

Debug output:

[ 0.003333] [00002ccd] libusb: debug [libusb_open] open 2.10
[ 0.003379] [00002ccd] libusb: debug [usbi_add_pollfd] add fd 28 events 4
[ 0.003394] [00002cd6] libusb: debug [handle_events] poll() returned 1
[ 0.003408] [00002cd6] libusb: debug [handle_events] caught a fish on the control pipe
[ 0.003419] [00002cd6] libusb: debug [libusb_try_lock_events] someone else is modifying poll fds
[ 0.003427] [00002cd6] libusb: debug [libusb_handle_events_timeout_completed] event handler was active but went away, retrying
[ 0.003432] [00002cd6] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.003436] [00002cd6] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.003443] [00002cd6] libusb: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 0.003847] [00002ccd] libusb: debug [libusb_claim_interface] interface 0
[ 0.003858] [00002ccd] libusb: debug [libusb_claim_interface] interface 0
/home/gunnar/dev/homepower/node_modules/usb/usb.js:284
	this.device.__claimInterface(this.id)
	            ^

Error: LIBUSB_ERROR_BUSY
    at Interface.claim (/home/gunnar/dev/homepower/node_modules/usb/usb.js:284:14)
    at Object.<anonymous> (/home/gunnar/dev/homepower/testusb2.js:18:20)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:754:12)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:622:3)

Help or ideas appreciated
Gunnar

kevinmehall, 2019-04-08:
`067b:2303` looks like a PL2303 USB-serial adapter? There will be a kernel serial driver attached to that device by default. You probably want to use `node-serialport` instead to use that kernel driver instead of talking to the device endpoints directly. But if you want to write your own driver for some reason, the `detachKernelDriver()` method will detach the kernel's driver and let you claim the interface.

closed 2019-04-08 by kevinmehall

gberth, 2019-04-10:
Thanks for your very informative answer. You were right all the way :-), so I'm on track again.