node-usb/node-usb #312: Build fail on Node.js 12

PaulBernier, 2019-05-15:
```bash
$ npm install usb

> node-hid@0.7.8 install /home/paul/factom/ledger-test/node_modules/node-hid
> prebuild-install || node-gyp rebuild

> usb@1.5.0 install /home/paul/factom/ledger-test/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp WARN Using needle for node-pre-gyp https download 
node-pre-gyp WARN Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.5.0/usb_bindings-v1.5.0-node-v72-linux-x64.tar.gz 
node-pre-gyp WARN Pre-built binaries not found for usb@1.5.0 and node@12.2.0 (node-v72 ABI, glibc) (falling back to source compile with node-gyp) 
make: Entering directory '/home/paul/factom/ledger-test/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
../src/node_usb.cc: In function ‘void Initialize(v8::Local<v8::Object>)’:
../src/node_usb.cc:66:84: warning: ‘bool v8::Object::Set(v8::Local<v8::Value>, v8::Local<v8::Value>)’ is deprecated: Use maybe version [-Wdeprecated-declarations]
  target->Set(Nan::New<String>("INIT_ERROR").ToLocalChecked(), Nan::New<Number>(res));
                                                                                    ^
In file included from /home/paul/.node-gyp/12.2.0/include/node/v8-internal.h:14:0,
                 from /home/paul/.node-gyp/12.2.0/include/node/v8.h:25,
                 from ../src/node_usb.h:12,
                 from ../src/node_usb.cc:1:
/home/paul/.node-gyp/12.2.0/include/node/v8.h:3359:26: note: declared here
                     bool Set(Local<Value> key, Local<Value> value));
                          ^
/home/paul/.node-gyp/12.2.0/include/node/v8config.h:322:3: note: in definition of macro ‘V8_DEPRECATE_SOON’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/node_usb.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE SetDebugLevel(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/node_usb.cc:99:73: error: no matching function for call to ‘v8::Value::Uint32Value()’
  if (info.Length() != 1 || !info[0]->IsUint32() || info[0]->Uint32Value() > 4) {
                                                                         ^
In file included from ../src/node_usb.h:12:0,
                 from ../src/node_usb.cc:1:
/home/paul/.node-gyp/12.2.0/include/node/v8.h:2567:41: note: candidate: v8::Maybe<unsigned int> v8::Value::Uint32Value(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT Maybe<uint32_t> Uint32Value(
                                         ^~~~~~~~~~~
/home/paul/.node-gyp/12.2.0/include/node/v8.h:2567:41: note:   candidate expects 1 argument, 0 provided
../src/node_usb.cc:103:53: error: no matching function for call to ‘v8::Value::Uint32Value()’
  libusb_set_debug(usb_context, info[0]->Uint32Value());
                                                     ^
In file included from ../src/node_usb.h:12:0,
                 from ../src/node_usb.cc:1:
/home/paul/.node-gyp/12.2.0/include/node/v8.h:2567:41: note: candidate: v8::Maybe<unsigned int> v8::Value::Uint32Value(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT Maybe<uint32_t> Uint32Value(
                                         ^~~~~~~~~~~
/home/paul/.node-gyp/12.2.0/include/node/v8.h:2567:41: note:   candidate expects 1 argument, 0 provided
../src/node_usb.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE GetDeviceList(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/node_usb.cc:116:35: warning: ‘bool v8::Object::Set(uint32_t, v8::Local<v8::Value>)’ is deprecated: Use maybe version [-Wdeprecated-declarations]
   arr->Set(i, Device::get(devs[i]));
                                   ^
In file included from /home/paul/.node-gyp/12.2.0/include/node/v8-internal.h:14:0,
                 from /home/paul/.node-gyp/12.2.0/include/node/v8.h:25,
                 from ../src/node_usb.h:12,
                 from ../src/node_usb.cc:1:
/home/paul/.node-gyp/12.2.0/include/node/v8.h:3368:26: note: declared here
                     bool Set(uint32_t index, Local<Value> value));
                          ^
[... 55 lines cut ...]
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/paul/factom/ledger-test/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:196:13)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:1011:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:268:5)
node-pre-gyp ERR! System Linux 4.15.0-48-generic
node-pre-gyp ERR! command "/home/paul/.nvm/versions/node/v12.2.0/bin/node" "/home/paul/factom/ledger-test/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/paul/factom/ledger-test/node_modules/usb
node-pre-gyp ERR! node -v v12.2.0
node-pre-gyp ERR! node-pre-gyp -v v0.11.0
node-pre-gyp ERR! not ok 
Failed to execute '/home/paul/.nvm/versions/node/v12.2.0/bin/node /home/paul/.nvm/versions/node/v12.2.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/paul/factom/ledger-test/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/paul/factom/ledger-test/node_modules/usb/src/binding --napi_version=4 --node_abi_napi=napi --napi_build_version=0 --node_napi_label=node-v72' (1)
npm WARN ledger-test@1.0.0 No description
npm WARN ledger-test@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb@1.5.0 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb@1.5.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/paul/.npm/_logs/2019-05-15T14_54_40_833Z-debug.log
```

kevinmehall, 2019-05-15:
Fixed on master by https://github.com/tessel/node-usb/pull/311, leaving this open to remind me to make a release.

PaulBernier, 2019-05-15:
Awesome!

Corben78, 2019-05-17:
As I'm hitting an issue with brother printers when using nodejs serialport's list() method, I'd like to try this package. Looking forward to release. Do you have any rough eta?

kevinmehall, 2019-05-19:
Released 1.6.0.

@Corben78 If your device is a USB serial port (not a vendor-class device that needs a custom driver), this probably isn't the library you're looking for, though.

closed 2019-05-19 by kevinmehall

Corben78, 2019-05-19:
Thank you very much!
Yeah I want to access an USB serial port eventually, but I also need to get the list of all attached USB/serial devices. When I got the name of the interface (e.g. /dev/ttyACM0 or COM3) I can use serialport for the rest. But just getting the list via serialport.list() is causing issues with the serialport library in combination with brother printers on Windows. Using this library was a suggestion from here: https://github.com/serialport/node-serialport/issues/1830#issuecomment-488536943

caffeinatedbits, 2021-09-29:
@Corben78 the Brother printer issue has now been resolved in node-serialport. See https://github.com/serialport/node-serialport/pull/2325