node-usb/node-usb #323: Data event does not give any date (ANT+ dongle)

JasperFioole21, 2019-08-17:
I use the following code to receive data from an ANT+ dongle I used other software (Zwift, which shows that the dongle actually received data):

```
/* Require the config */
require('dotenv').config()
/* require the usb class */
var usb = require('usb');

/* find the device */
var device = usb.findByIds(process.env.VENDOR_ID, process.env.PRODUCT_ID);
/* open the device */
device.open(true);
/* claim interface WHERE IS THIS FOR */
const [interface] = device.interfaces;
interface.claim();

/* Start Polling */
const [inEndpoint, outEndpoint] = interface.endpoints;
inEndpoint.startPoll();
// inEndpoint.transferType = 2;

inEndpoint.on('data', (d) => {
    console.log("usb data:", d, " len:", d.length)
});

inEndpoint.on('error', function (error) {
    console.log("on error", error);
});
```

Do I do something wrong?

kevinmehall, 2019-08-20:
Nothing looks wrong at the node-usb level, but I don't know what your device is expecting. You probably have to send the device a control transfer or write to another endpoint before it will start sending data on that endpoint.

closed 2019-08-20 by kevinmehall

JasperFioole21, 2019-08-20:
I'm using OS X, I've tested the USB with another program which keeps providing me with data (Heartrate strap), what could be the problem if it is not my code?

kevinmehall, 2019-08-20:
Have you sniffed the USB traffic from that other program to see what it does for initialization? You can't necessarily just read an endpoint and expect the device to send you what you want.

JasperFioole21, 2019-08-20:
I will try sniffing it, ant Ant+ dongle is a receiver to receive data from heart rate straps.

I've found a library (https://github.com/Loghorn/ant-plus) which is based in node-usb, but this one was not working either, thats why I started to look into the node-usb libraby. I'm starting to think dat I would net an other OS then OSX to get this to work.