node-usb/node-usb #326: Garbage Collection takes a long time

cinderblock, 2019-09-18:
I am using `node-usb` to transfer data to & from a custom USB device at ~500Hz (ideally up to 1kHz). One issue we're having is that at ~2Hz, the Node.js `gc()` (Garbage Collector) runs. Usually it's just a "Scavenge" type that takes ~9ms but often enough it's a "MarkSweepCompact" type that can easily take 50ms. This "glitch" in our runtime is problematic for our quasi-realtime system. (Yes, I know Node.js is not the first choice for a realtime system, but it's what I'm using for now).

In investigating this problem, our main issue comes from something relating to how we use `node-usb`. Once we remove reads & writes to our USB device, the time the GC takes drops by an order of magnitude (5ms peak). This indicates that it's either something in how `node-usb` works or something in my wrapping of `node-usb` that is giving the garbage collector extra work.

Looking at [my code](https://github.com/cinderblock/smooth-control/blob/master/src/SmoothControl.ts#L560), I'm using a couple `Promise`s & `async`/`await`s in my regular data handling path (some on read, some on write). I see that there have been some [recent improvements to V8](https://v8.dev/blog/fast-async) that should address this but for other reasons it's difficult for me to switch to Node 12 (which is when these improvements in V8 were incorporated in Node.js).

I'm going to explore how much the Promises are slowing things down but, for various reasons, I'm betting this is not the culprit. My suspicion is that it's how `node-usb` creates new `Buffer` objects for every read of data, or how I'm getting data out of those `Buffer`s. I'd try to mess with the `node-usb` API/internals first but I'm not super familiar with native modules.

Has anyone come across this kind of issue? Any thoughts on improving the memory usage here would be appreciated.

cinderblock, 2019-09-19:
Well, I decided to dive into it. I have pretty conclusive evidence that the way transfers are handled in the JavaScript part of this library is contributing significantly to extra work for the garbage collector.

Here is a plot of the achieved control period in my application with the current version of `node-usb`:

[image]

 - **Green** trace is direct plot of some of the observed periods. (Most dropped for performance reasons.)
 - **Red** trace is a variation of **Green** plot that captures recent peaks that would otherwise be missed and decays to zero slowly.
 - **Yellow** is the time V8 reports for each instance of garbage collection.
 - **Purple/White** show which type of GC was just run (Scavenge vs MarkSweepCompact).

Scale is number of seconds for each period. Max value is 50ms. Average of **Green** trace is ~2ms.

First notice that the GC events (when the **Yellow** line changes values, even slightly) correlate perfectly in time with the observed period peaks (**Red**). Also notice how long GC reported taking for the MarkSweepCompact (~40ms!) and how that correlated perfectly with **Red**'s overall peak.

Now take a look at what I achieved by manually doing what `dev.controlTransfer` does internally:

[image]

Nearly an order of magnitude reduction in all cases! The MarkSweepCompact events also seem to happen at a lower frequency.

The main thing that I changed to enable this was to not reallocate a new `usb.Transfer` on every call to transfer. Instead, I allocated a `Transfer` and a `Buffer` once, filled each with the correct data, and use that buffer with `transfer.submit`.

A simplified (and probably incomplete) example:

```JavaScript
const myDataLength = 37;

const transfer = new usb.Transfer(
  device,
  0,
  usb.LIBUSB_TRANSFER_TYPE_CONTROL,
  1000,
  (error, buf, actual) => {
    // Transfer complete
  }
);

// The full send buffer to pass to transfer.submit
const sendBuffer = Buffer.alloc(myDataLength + usb.LIBUSB_CONTROL_SETUP_SIZE);

// Part of the send buffer that we change regularly
const writeBuffer = sendBuffer.slice(usb.LIBUSB_CONTROL_SETUP_SIZE);

// ...

sendBuffer.writeUInt8(usb.LIBUSB_RECIPIENT_INTERFACE | usb.LIBUSB_REQUEST_TYPE_CLASS | usb.LIBUSB_ENDPOINT_OUT, 0);
sendBuffer.writeUInt8(0x09, 1);
sendBuffer.writeUInt16LE(0x0809, 2);
sendBuffer.writeUInt16LE(0, 4);
sendBuffer.writeUInt16LE(writeBuffer.length, 6);

// ...

function sendData(myData) {
  // Copy myData to writeBuffer
  transfer.submit(sendBuffer);
}
```

Note this leaves some things to be desired. The primary annoyance is that getting a "transfer complete" signal is a little convoluted.

cinderblock, 2019-09-19:
Similar issues exist on the IN transfer direction. I haven't captured nice pictures showing the difference but I'm pretty sure it's much better. My more efficient IN transfer:

```JavaScript
const inBuffer = Buffer.allocUnsafe(reportLength);

const inTransfer = endpoint.makeTransfer(
  1000,
  (error, buf, actual) => {
    if (error && error.errno != 4) {
      events.emit('error', error);

      return;
    }

    events.emit('data', parseHostDataIN(buf.slice(0, actual)));

    if (polling) doInTransfer();
  }
);

function doInTransfer() {
  inTransfer.submit(inBuffer);
}

if (polling) {
  doInTransfer();
}
```

thegecko, 2023-02-26:
@cinderblock Any recommendations (PR?) to resolve this?

cinderblock, 2023-02-26:
@thegecko All of what I did is captured in the comments above. I haven't looked at a more complete PR to "fix" node-usb's implementation. For my purposes, it was sufficient to just call the internal functions and move on.

Can I help clarify anything in the above implementations?