node-usb/node-usb #328: Why is ISerialNumber only one

GanZhiXiong, 2019-09-25:
[image]

kevinmehall, 2019-09-25:
It's the index of the string descriptor containing the serial number. Pass it to `getStringDescriptor` to fetch the actual serial number string from the device.

closed 2019-09-25 by kevinmehall

GanZhiXiong, 2019-09-25:
The following code I found from the other issue does open () before calling getStringDescriptor
Do you have to open the device to call getStringDescriptor？ 

```
device.open();
device.getStringDescriptor(deviceDescriptor.iManufacturer, function (err, manufacturer) {
  device.getStringDescriptor(deviceDescriptor.iProduct, function (err, product) {
    device.getStringDescriptor(deviceDescriptor.iSerialNumber, function (err, serialNumber) {
      console.log(manufacturer, product, serialNumber);
      device.close();
    });
  });
});
```

kevinmehall, 2019-09-25:
Yes

GanZhiXiong, 2019-09-25:
Because our device's idVenDor and idProduct are the same, I can't determine which device I'm connected to, unless I connect a device and then differentiate it based on the Serialnumber of the device, which becomes a bit cumbersome.