node-usb/node-usb #329: LIBUSB_ERROR_ACCESS

GanZhiXiong, 2019-09-26:
platform: macOS
The following code will report an error： LIBUSB_ERROR_NOT_FOUNDunde
```
outEndpoint.transfer(bytesToHex, function (e, d) {
        console.log('2' + e + d)

        // device.close()
    })
```

I found out from readme that I'm going to call claim() after I open it，
I then called claim after open, but reported the following error
```
Error: LIBUSB_ERROR_ACCESS
    at Interface.claim (/Users/ganzhixiong/Documents/typescript/Test/node_modules/usb/usb.js:280:14)
```

GanZhiXiong, 2019-09-26:
My complete code is as follows, I have been puzzled about this question for a long time. I am looking forward to your reply. Thank you!

``` ts
    var usb = require('usb');
    var device = usb.findByIds(0x1A86, 0xE0E0);
    device.open();
    var deviceInterface = device.interface(0);
    deviceInterface.claim();
    var outEndpoint = deviceInterface['endpoints'][1];
    var bytes = utils_1.hexToBytes("0260ff7319fa81ee1f87a900b810b0224c432255d5dfca1b6ca22fb1511982c887e87208cc1131be7bdf9113ad219798263d570415a93f782b91cce810fdfcdc");
    outEndpoint.transfer(utils_1.bytesToHex, function (e, d) {
        console.log('2' + e + d);
        // device.close()
    });
```

kevinmehall, 2019-09-26:
What is the device? On macOS you get LIBUSB_ERROR_ACCESS if the device is already claimed by a kernel driver. `detatchKernelDriver` only works on Linux. The only way to unbind a macOS kernel driver is to write a kernel extension.

GanZhiXiong, 2019-09-26:
This is a usb HID devices, using HIDSharp can open the equipment (https://www.zer7.com/software/hidsharp).

Equipment information is as follows：

Information for device HELLOKEY-30A31F8ED8F3BFCE (VID=0x1A86 PID=0xE0E0):

Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device address: 0x0004
Current configuration value: 0x01
Number of open pipes: 2

Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0110	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x1A86	idVendor
0xE0E0	idProduct
0x0000	bcdDevice
0x01	iManufacturer
0x02	iProduct
0x03	iSerialNumber
0x01	bNumConfigurations

Configuration Descriptor:
------------------------------
0x09	bLength
0x02	bDescriptorType
0x0029	wTotalLength   (41 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x04	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0x23	bMaxPower      (70 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x01	bInterfaceSubClass   
0x01	bInterfaceProtocol   
0x05	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0100	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0065	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x82	bEndpointAddress  (IN endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x18	bInterval         (24 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x18	bInterval         (24 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  
0x01   0x0000  Request failed with 0x0000001F
0x02   0x0000  Request failed with 0x0000001F
0x03   0x0000  Request failed with 0x0000001F
0x04   0x0000  Request failed with 0x0000001F
0x05   0x0000  Request failed with 0x0000001F

------------------------------

Connection path for device: 
Intel(R) 82801FB/FBM USB Universal Host Controller - 2658
Root Hub
Generic USB Hub
HELLOKEY-30A31F8ED8F3BFCE (VID=0x1A86 PID=0xE0E0) Port: 1

Running on: Windows 10 or greater (Build Version 18362)

Brought to you by TDD v2.12.0, Aug  2 2019, 16:35:05

GanZhiXiong, 2019-09-26:
With HIDSharp, you can communicate with your device on macOS and Windows, and everything is fine.And I've also been able to communicate with devices on Android.So I think it's the code, not the operating system

kevinmehall, 2019-09-26:
HID devices have a kernel driver, so you will not be able to use this library to communicate with them on macOS. You want to use a library based on libhid, not libusb, to use the kernel driver to communicate with the device.

closed 2019-09-26 by kevinmehall

GanZhiXiong, 2019-09-26:
Hello, can node-hid communicate with my device?
https://github.com/node-hid/node-hid/

Thank you very much for your reply

GanZhiXiong, 2019-09-26:
I've used node-hid, but I can't connect to my device!I have proposed an issue in node-hid. 
If you have time, please help me to have a look. This problem has been bothering me for a long time

https://github.com/node-hid/node-hid/issues/341

kevinmehall, 2019-09-26:
Looks like it's built on `hidapi` which is what you want.

The device details you've posted don't include the HID report descriptors that would identify whether it is a keyboard. On Linux you can run `lsusb -v -d 1A86:E0E0`, or the tool you are using may have the ability to show more detail on the HID descriptor.

If it works in hidsharp on macOS, I'd recommend looking at C code in the two libraries to compare what `IOHID` functions they use.

GanZhiXiong, 2019-09-26:
The operations you mentioned are a little difficult for me technically.I just want to find a library that I can use, and I don't have time to research any more.

Thank you very much for your reply!