node-usb/node-usb #332: Unable to detect USB devices plugged into a docking station

adamtLICOR, 2019-10-10:
I'm having some difficulty getting the `attach` event to fire when plugging a device in indirectly, specifically through this type of docking station:

https://www.dell.com/en-us/work/shop/accessories/apd/452-bcnp?mkwid=srusXMF3x

Does the fact the docking station uses Thunderbolt (USB-C) have anything to do with this issue? Should `node-usb` be able to detect USB-C events, or is that outside the project's scope?

Any help would be awesome!

kevinmehall, 2019-10-10:
https://github.com/libusb/libusb/issues/544

(though node-usb adds hotplug event support to libusb on Windows via a very old patch that probably makes the problem worse)

adamtLICOR, 2019-10-11:
Ah, thank you!

Reading through that thread now -- sounds like a known issue with `libusb`, then?

Do you want me to close out this **Issue**, or keep it open until `libusb` addresses **USB-c** support?

kevinmehall, 2019-10-11:
Let's close it. You'll also want to follow #161 about getting libusb updates into the prebuilt releases of this library.

closed 2019-10-11 by kevinmehall