node-usb/node-usb #333: Control Transfer Buffer checking is faulty

bayssmekanique, 2019-10-12:
The current implementation of `controlTransfer` relies on passing in a `Buffer` for and then uses the `Buffer.isBuffer` method for validation.

https://github.com/tessel/node-usb/blob/0d3c0e4694f52760c9b3f7ade0abb0f40ad50d03/usb.js#L104-L106

The problem is that the Node shim for `Buffer` calls a function that checks for `instanceOf`, which requires that the `Buffer` being passed in is utilizing the same library or else the validation fails, even if the implementations are compatible.  This leads to some very strange side-effects when working with systems that have different bundling processes as `node` targeted bundles will pull in a different shim than `web` targeted bundles and will now allow data to flow between the two.

A possible resolution is to include a `Buffer` implementation library.  This has the advantage of solving this initial problem because the `isBuffer` method in the [feross/buffer](https://github.com/feross/buffer) library is more robust and won't fail if the `Buffer` object is not using the same implementation library.  The other advantage is that this allows users of the library to pass in a `Buffer` of the exact same library implementation regardless of build platform and reduces the possibility of API mismatches.

The downside to this approach is an added dependency, which for a project with only 3 dependencies is a non-trivial increase.  The other potential solution would be to implement a more robust validation method similar to [feross/is-buffer](https://github.com/feross/is-buffer).

I will submit a PR with the added `Buffer` library attached to this issue for testing and review.

kevinmehall, 2019-10-12:
What if we made it check `buffer_or_length instanceof Uint8Array`, as both Node's buffer and Feross's buffer are subclasses of `Uint8Array`? To accept plain `Uint8Array`, the `data_or_length.copy(buf, SETUP_SIZE)` would have to be replaced with something like `buf.set(data_or_length, SETUP_SIZE)`.

bayssmekanique, 2019-10-13:
@kevinmehall, that seems like an excellent solution.  Are you able to make this change or would you like me to work on a PR?

kevinmehall, 2019-10-13:
Please go ahead and make a PR.

PR #342 (2019-11-17): Resolves Issue 333
Changes `data_or_length` property to allow any `Uint8Array` instance (including `Buffer`) rather than forcing node specific implementation.

Resolves #333
closed 2019-11-17 by kevinmehall