node-usb/node-usb #340: Add `ios` / `iphoneos` target + Add arm64 / arm64e / armv7 / armv7s architecture

brandonros, 2019-11-12:
Can you extend the node-gyp configuration to build a statically linked `.a` library for `arm64`? You can link against `/Applications/Xcode.app//Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS.sdk/System/Library/Frameworks/IOKit.framework` 

This will allow the compiled result to be used in iOS XCode projects

JamesHagerman, 2019-11-12:
This project was updated to use `prebuild` instead of `node-gyp` in #300

While I can not comment on iOS, it may be worth mentioning that I have started a branch which attempts to add `arm64` prebuild support to the existing Travis build: https://github.com/JamesHagerman/node-usb/tree/prebuild-for-arm64

However, my intent is to target Raspberry Pi specifically and I have not yet had a chance to test the compiled outputs on that platform.

brandonros, 2019-11-12:
Yeah, the tricky part is `arm64` for `darwin` and not `linux`. I appreciate the help. I guess I'm stuck trying to compile `libusb` for `iPhoneOS` in XCode, then linking the app against `Release/obj.target/usb.a`

It's all a proof of concept to see if iPhoneOS supports basic controlTransfer in/out and bulkTransfer when you do Thunderbolt -> USB C -> USB device. What do you think?