node-usb/node-usb #343: Cannot use node-usb with electron

GoogleFeud, 2019-11-17:
I get this error when I try running my main script with electron: `electron Engine/Engine.js`:

```
App threw an error during load
Error: A dynamic link library (DLL) initialization routine failed.
\\?\D:\Programs\Project\node_modules\usb\build\Release\usb_bindings.node
    at Error (native)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:583:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (internal/module.js:20:19)
    at bindings (D:\Programs\Pi-RetroConsole.js\node_modules\bindings\bindings.js:112:48)
```
Tried rebuilding the module, it said that the rebuild was successful, but the error still occurs. 

**Windows**: 10

**Dependencies**:
```
  "dependencies": {
    "electron": "^7.1.1",
    "usb": "^1.6.1",
    "ws": "^7.2.0"
  },
  "devDependencies": {
    "electron-rebuild": "^1.8.6"
  }
```

bayssmekanique, 2019-11-18:
I was receiving a similar error and it appears to be an issue with `electron-rebuild`.  For us, the solution was to switch to using `electron-builder` using the script `electron-builder install-app-deps` instead of the rebuild script.

quinton-ashley, 2020-10-20:
prebuild-install http 404 https://github.com/tessel/node-usb/releases/download/v1.6.3/usb-v1.6.3-electron-v82-win32-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=10.1.3 runtime=electron arch=x64 libc= platform=win32)

thegecko, 2021-11-08:
Closing due to inactivity, please reopen this issue if it persists with the latest release.

closed 2021-11-08 by thegecko