node-usb/node-usb #349: Can't use node-usb with electron react applicaiton.

karadagal, 2020-01-20:
Hello I'm trying to use this node-usb library for my electron react application but when I run the code inside the electron container, I get an error like this:

```
bindings.js:178 Uncaught TypeError: Cannot read property 'indexOf' of undefined
    at Function.t.getFileName (bindings.js:178)
    at t (bindings.js:82)
    at Object.<anonymous> (usb.js:1)
    at Object.<anonymous> (2.5fedd273.chunk.js:2)
    at l (index.html:1)
    at Module.66 (usbAccessButton.component.jsx:5)
    at l (index.html:1)
    at Object.41 (main.2cdae987.chunk.js:1)
    at l (index.html:1)
    at f (index.html:1)
```

```
import React from 'react';

import Button from 'react-bootstrap/Button';

var usb = require('usb');

class UsbAccessButton extends React.Component {

  render() {
      return <Button>Get USB Access</Button>
  }
}

export default UsbAccessButton;
```

```
import React from 'react';

import './App.css';

import UsbAccessButton from './components/usb-access-button/usbAccessButton.component';

function App() {
  return (
    <div className="App">
      <header className="App-header">
        <UsbAccessButton />
      </header>
    </div>
  );
}

export default App;
```

dioslibre, 2020-02-01:
Maybe you should use "electron-rebuild - f w usb" to rebuild the module for electon. And be sure to drop to electron 6. It has problems with native node modules.

BliTHE-LI, 2020-04-07:
How to solve ?
I have the same problem.

312977, 2020-04-24:
i  am  use electron vue ,has the same problem.

Blackfaded, 2020-07-28:
```sh
npm install electron-rebuild --dev
./node_modules/.bin/electron-rebuild
```

worked for me in main process

petertorelli, 2020-10-17:
The problem is that `bindings` searches the stack trace to find the `usb_bindings.node` file, and the stack trace does not contain a callsite for `./node_modules/usb/build/Release/usb_bindings.node`. I have an issue opened on this, would be great to get an assist on this with someone more familiar with why `bindings` is used...

https://github.com/TooTallNate/node-bindings/issues/29

blumk, 2020-12-29:
Here's a solution that finally worked for me:

Generating the `usb_bindings.node` with `electron-rebuild` is just the first step.

The error message comes from the file system trying to access the bindings file:
`Cannot read property 'indexOf' of undefined at Function.t.getFileName (bindings.js:178)`

**Explanation**

> since the renderer is served as web page, potentially from an in-memory file system (so the *.node file may not be emitted to disk, and Electron will never see it)

**Solution**
> Second, we must substitute node-bindings with our own shim, src/bindings.js:

```
module.exports = x =>
  __non_webpack_require__(
    `${require("electron").remote.app.getAppPath()}/${x}`
  );
  
```

Full explanation and tutorial:  https://stackoverflow.com/a/50547650

mttjcksn, 2021-04-06:
Hi all, 

I'm trying to use this (or indeed any!) usb library within my typescript reactjs electron app.  Using node-usb as described in this project's readme, I get the following error:

`TypeError: exists is not a function.    Function.getRoot (node_modules/bindings/bindings.js:202)`

I'm also not using webpack.

I have run `electron-rebuild - f w usb `, and added a `bindings.js` to `node_modules\usb\src` with the contents described above.  The error remains the same.

Any help would be very much appreciated.

0xMars4204, 2021-07-15:
@mttjcksn, I have the same issue with you

```TypeError: exists is not a function. Function.getRoot (node_modules/bindings/bindings.js:202)```

Did you find out the solution?

Any help would be appreciated.
Thanks in advance.

thegecko, 2021-07-15:
Please ensure you are running `node-usb` in the backend process of Electron

0xMars4204, 2021-07-15:
> Please ensure you are running `node-usb` in the backend process of Electron

How to run `node-usb` in the backend process?
`getRoot` function always trying to find `http://` instead of `file://.

thegecko, 2021-11-08:
Closing due to inactivity, please reopen this issue if it persists with the latest release.

closed 2021-11-08 by thegecko