node-usb/node-usb #350: NODE_MODULE_VERSION error.

Stanlyhalo, 2020-02-13:
So I am using electron, and nodejs is updated to the latest v13.8.0. This is what I get. I can't find any nodejs version that is 75. How can I resolve this?

App threw an error during load
Error: The module '\\?\D:\axiom-drive\software\szechuan\node_modules\usb\build\Release\usb_bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 79. This version of Node.js requires
NODE_MODULE_VERSION 75. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func (electron/js2c/asar.js:140:31)
    at process.func [as dlopen] (electron/js2c/asar.js:140:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:922:18)
    at Object.func (electron/js2c/asar.js:140:31)
    at Object.func [as .node] (electron/js2c/asar.js:140:31)
    at Module.load (internal/modules/cjs/loader.js:735:32)
    at Module._load (internal/modules/cjs/loader.js:648:12)
    at Module._load (electron/js2c/asar.js:717:26)
    at Function.Module._load (electron/js2c/asar.js:717:26)
    at Module.require (internal/modules/cjs/loader.js:775:19)
npm ERR! code ELIFECYCLE
npm ERR! errno 3221225477
npm ERR! szechuan@1.0.8 start: `electron .`
npm ERR! Exit status 3221225477
npm ERR!
npm ERR! Failed at the szechuan@1.0.8 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\DontT\AppData\Roaming\npm-cache\_logs\2020-02-13T03_37_11_339Z-debug.log

mreis1, 2020-02-17:
This is because your forgot to `npm install` your modules that require nativeaddons after updating your node installation.  Run npm install usb on your project and you should be ok

pythoulon, 2020-02-18:
@mreis Not sure. It turns out I had the same problem, with node-usb as a dependency of @abandonware/noble (see [my original post here](https://github.com/abandonware/noble/issues/42)).

`npm install` just wouldn't cut it. I had to remove all dependencies (including clearing the npm cache), install electron and electron-rebuild, reinstall all dependencies, then run electron-rebuild against electron v6 (in my case, it was NODE_MODULE_VERSION 73, the one used by electron 6), and only then did it work. I would have expected the install process to do this for me... (should it ?).

Stanlyhalo, 2020-02-26:
This still didn't fix it

matextrem, 2020-03-27:
Any update? I have the same problem.

quinton-ashley, 2020-09-01:
Any update now?

vipullimbachiya, 2020-09-09:
> Any update now?

`electron-builder install-app-deps`

For me, this worked

meghajayakumar, 2020-09-23:
Having the same issue, any update?

pythoulon, 2020-09-23:
VSCode keeps changing the Electron version it uses and expects. See my comment there: https://github.com/abandonware/noble/issues/42#issuecomment-697572629

ngvcanh, 2020-10-17:
I had the same problem and I found a solution.
Delete the `node_modules` folder and run commands follow in `cmd`, don't run at vscode terminal or git cmd:

```
npm install
npm install usb --save
npm install electron-rebuild --save-dev
node_modules\.bin\electron-rebuild
```

It's working for me

quinton-ashley, 2020-10-18:
It works for me, I think they just fixed it! I still have electron-rebuild as a global pkg. Didn't change anything I was doing.

quinton-ashley, 2020-10-19:
Actually it only worked on macOS. On windows there are too many errors in one of the files for the project to build.
```
C:\Users\quint\AppData\Local\node-gyp\Cache\10.1.3\include\node\v8.h(9918): fatal error C1003: error count exceeds 100; stopping compilation (compiling source file ..\src\device.cc) [E:\dev\apps\nostlan\node_modules\usb\build\usb_bindings.vcxproj]
```

quinton-ashley, 2020-10-20:
I'm using VS 2019. Why are there so many errors?

thegecko, 2021-10-18:
This simple example may help you:

https://github.com/node-usb/node-usb-example-electron

thegecko, 2021-11-08:
Closing due to inactivity, please reopen this issue if it persists with the latest release.

closed 2021-11-08 by thegecko