node-usb/node-usb #351: No Error, No Timeout

PGP-Protector, 2020-02-18:
I have a script that's scanning my connected USB Devices to find printers, works except for when it comes across my VR Headset.

Calling getStringDescriptor via

    getManufacturerString (counter, idVendor, idProduct, index)
    {
        var term = usb.findByIds(idVendor, idProduct);
        term.open();
	
        try
        {
            term.getStringDescriptor(index, counter.callbackClosure(counter, counter.myManufacturerStringCallback));
        }
        catch(err)
        {
            console.log(err);
        }
    }

With a VR Headset (idVendor = 1118. idProduct=1625) 
The call term.getStringDescriptor never returns.  No error is thrown, just goes into limbo.

term.open() doesn't throw any error (tried that)

kevinmehall, 2020-02-23:
Where are you getting the `index` passed to this function? Generally you would get it from the device descriptor, not know it ahead of time.

What OS is this on? 

Can you provide a libusb debug log by setting the `LIBUSB_DEBUG=4` environment variable?

thegecko, 2021-11-08:
Closing due to inactivity, please reopen this issue if it persists with the latest release.

closed 2021-11-08 by thegecko