node-usb/node-usb #353: node-usb not found for electron 8

sgoll, 2020-04-09:
The problem occurs when building the native dependency. The essential error message is this:

```
prebuild-install http 404 https://github.com/tessel/node-usb/releases/download/v1.6.2/usb-v1.6.2-electron-v76-win32-x64.tar.gz
```

There is no release of `node-usb@1.6.2` for `electron-v76`, as can be confirmed over at the [releases page](https://github.com/tessel/node-usb/releases/tag/v1.6.2).

@kevinmehall Would it be possible to issue a release that supports `electron-v76`? When I look at the corresponding commit dce21022559e486f6e1245adea927bbd3d5b0454 for electron 6 and 7, it looks like editing `.travis.yml` might be enough, i.e. adding `8.0.0` to `PREBUILD_ELECTRON_TARGETS`.

(Mentioning serialport/node-serialport#2031 for similarity)

kevinmehall, 2020-04-11:
Yes, if it's supported in https://github.com/lgeiger/node-abi/ (used by prebuild), then you can just submit a PR adding it, and when I tag a release it will be built automatically.

rajivshah3, 2020-04-12:
PR: #354

PR #354 (2020-04-12): Add prebuild for Electron 8
Adds prebuild for Electron 8 (https://github.com/lgeiger/node-abi/blob/14ee3eb65cdd51cc0dd0b7fa3220b07451a02ef9/index.js#L76)

Fixes #353﻿
closed 2020-04-12 by kevinmehall