node-usb/node-usb #355: No Error, No Timeout，Too

apemen, 2020-04-21:
`var usb = require('usb')
console.log("22222222")
var device = usb.findByIds(0x1ad4,0x00c8)
if(device){
  device.open();
  var interface = device.interfaces[0];
  interface.claim();

  var inpoint = interface.endpoints[0]; // interface.endpoint(id) don't work for id=0 btw
  var outpoint = interface.endpoints[1];

}
console.log(device)`

use “electron .”

but ,i got nothing

apemen, 2020-04-21:
then i find there is no "usb.getDeviceList()" function in usb.js

where is wrong

kevinmehall, 2020-04-21:
The getDeviceList function is [defined in C++](https://github.com/tessel/node-usb/blob/master/src/node_usb.cc#L107). Endpoint address 0 is the control endpoint -- use `device.controlTransfer` to make control transfer requests. Your code is not actually issuing any transfers. Do you know which call is failing? With Electron, make sure you've built the library against the correct Electron ABI version.

apemen, 2020-04-22:
in my code
 “var device = usb.findByIds(0x1ad4,0x00c8)
console.log("22222222")
”
 this row can‘t get device info . and the next row will't be print out

thegecko, 2021-11-08:
Closing due to inactivity, please reopen this issue if it persists with the latest release.

closed 2021-11-08 by thegecko