node-usb/node-usb #357: LIBUSB_ERROR_NOT_SUPPORTED on windows 10 even after installing WinUSB driver with Zadig

snuffysasa, 2020-05-03:
First of all, this is working great for me on linux (ubuntu), and thank you! 

Unfortunately, I am having trouble getting it to work on windows 10. I used the program Zadig as recommended, and was able to find the devices listed with the vendor id that I am using and have been able to install drivers for them.  (I had some trouble related to disable driver signing on windows), but eventually go it.  I use the driver: WinUSB (v6.1.7600.16385), because that is what it seemed like the ReadMe was recommending.  After successfully installing and rebooting, I still get the same LIBUSB_ERROR_NOT_SUPPORTED error when trying to run my nodejs program.  I also tried the over available drivers from Zadig, but still no luck.

Any ideas what to do?

Thanks!

snuffysasa, 2020-05-03:
Update: I solved this issue.  I read in another issue page about the detachKernelDriver is not necessary on windows and breaks the program.

snuffysasa, 2020-05-03:
Additionally: To use on the latest version of windows 10, these steps are necessary before using Zadig to install the driver:  [Disable driver Signing](https://jbcomp.com/disable-device-driver-signing-in-windows-10/)

closed 2020-05-03 by snuffysasa

kevinmehall, 2020-05-04:
Yes, `detachKernelDriver` always reports LIBUSB_ERROR_NOT_SUPPORTED on any OS besides Linux.

Theoretically the WinUSB driver should be signed by Microsoft -- you're just binding a generic driver that already comes with Windows > 7 to the device, so I'm not sure why disabling driver signing would be necessary, but I haven't tested it with recent versions of Windows 10.

Chris2011, 2020-10-31:
I also did the driver signature disabling, but still got the same error. I did this with Zadig:

* Open Zadig
* Go to options -> List all devices
* Select the usb flash drive in the DropDownListBox
* See the driver, which is installed for that device

For me it was a different one and I've overwritten it with WinUSB. After this, I don't get the error anymore, just undefined, but this could be from my code. Maybe didn't open the connection to the device, dunno. But maybe it helps someone.

The problem was a different driver for my USB Stick.