node-usb/node-usb #361: node-usb not found for electron 9

alejandroclaro, 2020-05-25:
Please add prebuild for electron 9.

It's alredy supported by node-abi v2.17.0.

alejandroclaro, 2020-07-13:
Any news?

Looks like someone adds the prebuild generation in Travis, but is not working properly.

```
prebuild-install http 404 https://github.com/tessel/node-usb/releases/download/v1.6.3/usb-v1.6.3-electron-v80-linux-x64.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=9.0.5 runtime=electron arch=x64 libc= platform=linux)
```

sgoll, 2020-09-04:
FWIW, the same applies to Electron 10 which has been released in the meantime:

```
    prebuild-install http 404 https://github.com/tessel/node-usb/releases/download/v1.6.3/usb-v1.6.3-electron-v82-linux-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=10.1.1 runtime=electron arch=x64 libc= platform=linux)
```

thegecko, 2021-01-30:
This has been added and just needs a new release AFAICT.

cc @kevinmehall

kevinmehall, 2021-01-30:
Thanks to @thegecko's help, v1.6.4 has been released with prebuilds for Node 14 and Electron 9, 10, 11, and 12-beta1.

closed 2021-01-30 by kevinmehall