node-usb/node-usb #365: Low performance on Win10 with USB3 devices?

petertorelli, 2020-06-06:
I have an ElectronJS application I run on the same laptop in either a Win10 or Ubuntu 18.04 partition. The hardware is a Joulescope JS110 power monitor that sends back 2M samples/sec (four byte samples). I'm using endpoint `startPoll` and `endpoint.on('data')`. 

I moved the code to a forked child process to avoid throtting Electron.

On Win10 it completely throttles the forked child process to the point where a debug setTimeout doesn't even respond. 

I've experimented with the number of transfers and packet size and can decrease throttling but increasing the number of transfers, but then data is dropped (each packet from the device has a sequence counter in it, so I can keep track of how many I've lost).

Here's the weird part: the same code works fine on the same platform running Ubuntu 18.04. It also works fine on more powerful Win10 machines.

Thoughts?

electron: 5.0.12
usb: 1.6.0

mildsunrise, 2022-03-25:
this looks like a loop starvation problem, see #383.
if you're on windows, could you try with the latest v2?
the loop should behave more fairly (or at the very least, not become completely unresponsive).

petertorelli, 2022-03-25:
@mildsunrise I'll give it a try and report back, it'll take me about a week to get setup to do this. The device in question is a USB3.0 voltage & current meter that sends a constant 32MB/s stream of data, so that probably qualifies as "If USB events are arriving fast enough".

mildsunrise, 2022-03-25:
yeah, that qualifies :) it also depends on the amount of work you do immediately in the transfer completion callback.
I'm not 100% sure this is a loop starvation problem, and also not 100% sure loop starvation is now solved on Windows, so don't hold your breath on it, but testing can't hurt.

thegecko, 2022-03-25:
Fixes now in usb@2.2.0

mildsunrise, 2022-03-26:
btw, you may also want to check out my implementation of `startPoll`: https://github.com/mildsunrise/hackrf.js/blob/f786db1959767bdc1e241f3544c83e1250bf2121/lib/interface.ts#L36

it defers work to `setImmediate` in order to keep the loop responsive to other events. even if v2 improves things, it's always a good idea to do something like this in your code.

I've successfully used it to stream 40MB/s of samples from an SDR while keeping things responsive.

petertorelli, 2023-01-20:
I've left the company working on this project. I'll leave it up to the new owner to resolve the issue. Thanks for the suggestions!

closed 2023-01-20 by petertorelli