node-usb/node-usb #366: Install fails on Raspberry Pi Zero with Node 10.20.1

InZan3, 2020-06-09:
`> usb@1.6.3 install `/home/pi/node-> usb@1.6.3 install /home/pi/node-usb
> prebuild-install --verbose || node-gyp rebuild

prebuild-install info begin Prebuild-install version 5.3.3
prebuild-install info install installing standalone, skipping download.
gyp ERR! clean error 
gyp ERR! stack Error: EACCES: permission denied, unlink 'build/Makefile'
gyp ERR! System Linux 4.19.118+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node-usb
gyp ERR! node -v v10.20.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb@1.6.3 install: `prebuild-install --verbose || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb@1.6.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-06-09T13_55_09_804Z-debug.logusb`
> prebuild-install --verbose || node-gyp rebuild

prebuild-install info begin Prebuild-install version 5.3.3
prebuild-install info install installing standalone, skipping download.
gyp ERR! clean error 
gyp ERR! stack Error: EACCES: permission denied, unlink 'build/Makefile'
gyp ERR! System Linux 4.19.118+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node-usb
gyp ERR! node -v v10.20.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb@1.6.3 install: `prebuild-install --verbose || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb@1.6.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-06-09T13_55_09_804Z-debug.log`

ferhattelci, 2020-07-04:
Hi, 

have you found a solution or does the problem persist? Because I can't install the package on my Zero too...

InZan3, 2020-07-05:
Hi, i dont need it anymore ;-)

kevinmehall, 2021-01-30:
This looks like you have mixed file ownership in your build or node_modules directory. To avoid this, don't run `npm` with `sudo`.

closed 2021-01-30 by kevinmehall