node-usb/node-usb #367: no suitable image found

saifeiLee, 2020-06-16:
[image]

Could anyone please explain why it happens?

kevinmehall, 2020-06-16:
Same as https://github.com/tessel/node-usb/issues/360 except now you've got a Linux binary on macOS. Are you sharing an install directory between operating systems?

closed 2020-06-16 by kevinmehall

saifeiLee, 2020-06-17:
> Same as #360 except now you've got a Linux binary on macOS. Are you sharing an install directory between operating systems?

Thanks for replying.
I only run my app on osx. No other system.But I package my app for linux by electron-builder. Is the packaging for linux causing the issue？

kevinmehall, 2020-06-18:
I've never used electron-builder, no idea.

saifeiLee, 2020-06-19:
> I've never used electron-builder, no idea.

Thank you all the same. May you have a lovely day~