node-usb/node-usb #368: Windows 7 64-bit node.js v10, v12, v14 0xC0000005 / -1073741819 error

brandonros, 2020-07-01:
```
C:\Users\Brandon\Documents\GitHub\node-tactrix-openport>node test.js
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------

[ 0.153009] [00000468] libusb: debug [libusb_get_device_list]
[ 0.154009] [00000468] libusb: debug [libusb_get_device_descriptor]
[ 0.154009] [00000468] libusb: debug [libusb_get_device_descriptor]
[ 0.154009] [00000468] libusb: debug [libusb_get_device_descriptor]
[ 0.154009] [00000468] libusb: debug [libusb_get_device_descriptor]
[ 0.154009] [00000468] libusb: debug [libusb_get_device_descriptor]
[ 0.154009] [00000468] libusb: debug [libusb_get_device_descriptor]
[ 0.154009] [00000468] libusb: debug [libusb_get_device_descriptor]
[ 0.154009] [00000468] libusb: debug [libusb_get_device_descriptor]
[ 0.156009] [00000468] libusb: debug [libusb_get_device_descriptor]
[ 0.156009] [00000468] libusb: debug [libusb_get_config_descriptor] index 0

C:\Users\Brandon\Documents\GitHub\node-tactrix-openport>echo %errorlevel%
-1073741819
```

Seems to fail right after `usb.getDeviceList()`. I've tried every version of node.js. Not sure what else to try

kevinmehall, 2020-07-07:
That's a segfault in the C or C++ code; next step would be to get a backtrace with a debugger.

You could try [building against a newer version of libusb](https://github.com/tessel/node-usb/pull/345) (assuming you don't need hotplug functionality) in case it's a bug fixed there.

brandonros, 2020-07-24:
this was related to the `webusb` package https://github.com/thegecko/webusb/issues/68

closed 2020-07-24 by brandonros