node-usb/node-usb #370: HID device, read report descriptor

AdamChlupacek, 2020-07-07:
Hello,

As said in the title, I am trying to read report descriptor from a HID device so that we can implement support for generic headphones control inside of our application.

I have managed to do this via `controlTransfer` on macOS, but on Windows I required the WinUSB driver as I found out later from the docs. I cannot make every of our users to manually download the WinUSB driver for their device.

Is there any other way how would I be able to get the HID Report Descriptor from a device on Windows? I have read through various HID related APIs on windows but did not find anything that would give me a HID Report Descriptor.

kevinmehall, 2020-07-07:
WinUSB ships with Windows 7 and later. If you control the device firmware, you can make Windows automatically bind it to your device by adding [WCID descriptors](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/automatic-installation-of-winusb). Otherwise, you can bind the driver with the [libwdi](https://github.com/pbatard/libwdi) library Zadig uses as its backend.

However, if your device is compatible with other HID functionality (rather than just using HID as a transport for arbitrary data), you may not want to use WinUSB since it overrides the default Windows HID driver. You'd probably be better off with node-hid or a similar library that works on top of the OS HID stack rather than replacing it. I don't know if it can read descriptors though. I think it assumes you know the report format ahead of time.

closed 2020-07-07 by kevinmehall

AdamChlupacek, 2020-07-07:
Thank you for your answer, 

The issue is that the devices are not ours, its just generic headphones with buttons. 

We are already using node-hid, but as you said, there seems to be no way how to get descriptors. I guess for now I will be dumping the descriptors on my macOS machine and hardcoding them in the application until there will be some other way how to get the HID descriptors on windows.

Thank you for your help.