node-usb/node-usb #371: Offline installation

eranl99, 2020-07-08:
I've tried to download usb with npm -i and yarn and adding the tgz file to a local offline Verdaccio registry.

Unfortunately, when I try to install it in an offline computer it seems like the installation script tries to reach the repository for some reason.

Why isn't the tgz file enough for the installation? Why do I need to have an internet connection?

kevinmehall, 2020-07-08:
[prebuild-install](https://github.com/prebuild/prebuild-install) downloads a precompiled binary of libusb and the node extension for your platform. Pass `--build-from-source` to skip this and build from the source in the tarball. You'll need a C++ toolchain installed.

closed 2020-07-08 by kevinmehall

eranl99, 2020-07-13:
Thanks.
When I do npm install --build-from-source=usb I still get an error. It seems like the flag is not passed on to the actual installation of usb.
I get an error (this is extracted from the verbose log):
Error: usb@1.6.3 install: `prebuild-install --verbose || node-gyp rebuild`

The actual installation just shows that it received an Error: getaddrinfo EAI_AGAIN nodejs.org at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:56:26).

kevinmehall, 2020-07-13:
That may be node-gyp downloading C++ headers in order to perform the build. node-usb prebuilds are on github.com, not nodejs.org.