node-usb/node-usb #378: Fails to run in a worker_thread

brandonros, 2020-08-28:
```
FATAL ERROR: HandleScope::HandleScope Entering the V8 API without proper locking in place
 1: 0x1010285f9 node::Abort() (.cold.1) [/Users/brandonros/.nvm/versions/node/v12.16.3/bin/node]
 2: 0x10008634d node::FatalError(char const*, char const*) [/Users/brandonros/.nvm/versions/node/v12.16.3/bin/node]
 3: 0x10008648e node::OnFatalError(char const*, char const*) [/Users/brandonros/.nvm/versions/node/v12.16.3/bin/node]
 4: 0x10018a6a1 v8::HandleScope::HandleScope(v8::Isolate*) [/Users/brandonros/.nvm/versions/node/v12.16.3/bin/node]
 5: 0x1000015cc node::InternalCallbackScope::InternalCallbackScope(node::Environment*, v8::Local<v8::Object>, node::async_context const&, int) [/Users/brandonros/.nvm/versions/node/v12.16.3/bin/node]
 6: 0x100001a70 node::InternalMakeCallback(node::Environment*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*, node::async_context) [/Users/brandonros/.nvm/versions/node/v12.16.3/bin/node]
 7: 0x100001d1e node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*, node::async_context) [/Users/brandonros/.nvm/versions/node/v12.16.3/bin/node]
 8: 0x102f8d37f handleCompletion(Transfer*) [/Users/brandonros/Desktop/uds-server/node_modules/usb/build/Release/usb_bindings.node]
 9: 0x102f8e48b UVQueue<Transfer*>::internal_callback(uv_async_s*) [/Users/brandonros/Desktop/uds-server/node_modules/usb/build/Release/usb_bindings.node]
10: 0x1008e840d uv__async_io [/Users/brandonros/.nvm/versions/node/v12.16.3/bin/node]
11: 0x1008f7c12 uv__io_poll [/Users/brandonros/.nvm/versions/node/v12.16.3/bin/node]
12: 0x1008e8846 uv_run [/Users/brandonros/.nvm/versions/node/v12.16.3/bin/node]
13: 0x1000bb7a2 node::NodeMainInstance::Run() [/Users/brandonros/.nvm/versions/node/v12.16.3/bin/node]
14: 0x100062e9f node::Start(int, char**) [/Users/brandonros/.nvm/versions/node/v12.16.3/bin/node]
15: 0x7fff6a59bcc9 start [/usr/lib/system/libdyld.dylib]
16: 0x2 
Abort trap: 6
```

device-worker.js

```
const Device = require('./lib/gs_usb')

const run = async () => {
  const device = new Device()
  await device.init()
  device.recv()
}

run()
```

main.js

```
const { Worker, isMainThread, parentPort, workerData } = require('worker_threads')
if (isMainThread) {
  const worker = new Worker('./device-worker.js')
  worker.on('error', (err) => console.error(err))
  worker.on('exit', () => console.error('exit'))
  console.log(worker)
}
```

mildsunrise, 2022-05-01:
It looks like this module is finally going to support use in worker threads, see #512. If you want, you can test that branch with your use case now.

thegecko, 2022-05-07:
Please check this issue still arises with the latest release; v2.4.0

closed 2023-02-26 by thegecko