node-usb/node-usb #381: Current API is unsafe

mildsunrise, 2020-09-05:
The current API has some design problems... the most important one is that it relies on the active configuration to not change. [According to libusb](http://libusb.sourceforge.net/api-1.0/libusb_caveats.html#configsel), this is only guaranteed once an interface is claimed. This causes, among other things:

 - Race: If another program changes the configuration after `device.open()` is called (but before the interface is claimed), then `configDescriptor`, `interfaces` and endpoint info will be invalid.

 - No way to [select another configuration](http://libusb.sourceforge.net/api-1.0/libusb_caveats.html#configsel) according to the procedure.

 - No way to check the active configuration, since `configDescriptor` is cached on first access.

 - No way to access interfaces (i.e. to detach kernel drivers, or claim them) without setting a configuration first (potentially causing an unnecessary reset).

#377 attempts to mitigate some of them, but we'd need to redesign the API for a proper fix.

In the meantime, we should clarify the situation in the README. The current phrase ("Configurations other than the default one" are not supported) is vague and insufficient IMHO.

thegecko, 2023-02-26:
Any progress on this @mildsunrise ?

cc @jreineckearm

mildsunrise, 2023-02-27:
a redesign of the legacy API is likely not going to happen. I think these unsafety problems are not present (or are at least less present) in the WebUSB API, but I have to check.

if the WebUSB API fixes those problems, the best way to go IMHO would be to rewrite the WebUSB adapter so that it uses the native bindings directly, avoiding this unsafety.

thegecko, 2023-02-27:
> if the WebUSB API fixes those problems, the best way to go IMHO would be to rewrite the WebUSB adapter so that it uses the native bindings directly, avoiding this unsafety.

That's an interesting approach